ssieb/esphome_components #1: CWWW2 compile error "name 'CONF_GAMMA_CORRECT' is not defined"

patfelst, 2020-01-31:
Hi there, thanks for writing this. I'm trying it on a [Brilliantsmart Trilogy CWWW](https://www.brilliantsmart.com.au/smart-products/lighting/smart-trilogy/) downlight. I already have a sort-of-working ESPHome config based on [this](https://esphome.io/cookbook/mirabella-genio-bulb.html#cold-warm-white-bulbs), but the light goes out when the brightness slider is reduced to less than about 40% and the colour temp slider only works on the right hand half. It's very warm on the right hand side, and cool (blue) in the middle of the slider, but it also dims a lot on the blue end of the scale (middle). If I reduce colour temp below half way, the light goes off.

**Anyway, my problem with your code is** I wasn't sure how to include it in ESPHome, I copied the whole directory "cwww2" to my Hassio `\config\esphome\custom_components\cwww2` and then included it like this:

```
esphome:
  name: brilliantsmart_downlight_cct_1
  platform: ESP8266
  board: esp01_1m
  includes:
    - custom_components/cwww2/cwww2_light_output.h
```
But I get an error related to CONF_GAMMA_CORRECT

```
INFO Reading configuration /config/esphome/test.yaml...
ERROR Unable to load custom component cwww2.light:
Traceback (most recent call last):
  File "/opt/esphome/esphome/config.py", line 140, in _lookup_module
    module = importlib.import_module('custom_components.{}'.format(domain))
  File "/usr/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/config/esphome/custom_components/cwww2/light.py", line 17, in <module>
    cv.Optional(CONF_GAMMA_CORRECT, default=1.0): cv.positive_float,
NameError: name 'CONF_GAMMA_CORRECT' is not defined
Failed config
```

ssieb, 2020-01-31:
I should update the README to explain.  You can clone this repo right into your config directory. Otherwise you will have to create a directory called `custom_components` there and copy the entire cwww2 directory into that.

ssieb, 2020-01-31:
You don't use any includes.  It is just like any other normal component.  Here's the config that he posted:
```yaml
light:
  - platform: cwww2
    id: mylight
    name: "test cwww2"
    color_temperature: output2
    brightness: output1
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2700 K

```

patfelst, 2020-02-01:
> Otherwise you will have to create a directory called `custom_components` there and copy the entire cwww2 directory into that.

Yes that's what I did, so this is what I have:
[image]

And I've now removed the "Includes:" statement, however I'm still getting the compile error "NameError: name 'CONF_GAMMA_CORRECT' is not defined". Did you forget to import that as it doesn't appear in this line

```
from esphome.const import CONF_OUTPUT_ID, CONF_COLOR_TEMPERATURE, CONF_BRIGHTNESS, \
    CONF_COLD_WHITE_COLOR_TEMPERATURE, CONF_WARM_WHITE_COLOR_TEMPERATURE
```

ssieb, 2020-02-01:
You're right, I added that override later, but forgot the import.  I've made an update.

patfelst, 2020-02-01:
Success! It works perfectly, the dimmer slider in Hassio front end works perfectly, as does the colour temperature slider - thanks so much!

closed 2020-02-01 by ssieb