ssieb/esphome_components #2: Bit mapped bytes need to be swapped

patfelst, 2020-03-07:
Thanks again for developing this custom component. I got around to assembling my circuit and testing today. After seeing garbled characters, but with the correct number of characters in the correct digit position, I found that the order of the 8-bit bytes in the 16-bit number of the bitmap need to be swapped in this line:
https://github.com/ssieb/custom_components/blob/3067cf945198ee357232269aa0fe35ddbf1a68cf/lcd_ht16k33/ht16k33_display.cpp#L71

I replaced that line with this:
`this->buffer_[pos] = (fontc << 8) | (fontc >> 8);`

And that fixed it! This may not be the most efficient way of swapping the bytes around, and maybe it would be better to do it in the i2C writes, but I couldn't find that code - so feel free to improve.

The **brightness** works too, after I realised that I have to enter it as a float between 00. and 1.0! Might be better to have it as an integer between 0 and 15?

I also added some extra code to **include decimal point**. The entire function that works for me is:
```
void HT16K33LCDDisplay::print(const char *str) {
  uint8_t pos = 0;
  uint16_t fontc = 0;
  for (; *str != '\0'; str++) {
    if (pos >= 4) {
      ESP_LOGW(TAG, "writing off the screen!");
      break;
    }

    uint8_t c = *reinterpret_cast<const uint8_t *>(str);
    if (c > 127)
      fontc = 0;
    else
      fontc = pgm_read_word(&alphafonttable[c]);
    this->buffer_[pos] = (fontc << 8) | (fontc >> 8);
    //ESP_LOGW(TAG, "pos[%u]=%c bitmap=%#04x", pos, c, fontc);
    
    // Look ahead to next character, if decimal point, add it to the current digit
    uint8_t next_char = str[1];
    if ((next_char != '\0') && (next_char == '.')) {
      // set the decimal point segment (bit-14) in current digit
      this->buffer_[pos] |= 0b0000000001000000;
      // discard the decimal point
      str++;
      if (*str == '\0')
        // break in the unlikely event the string ends with a decimal point!
        break;
    }
    pos++;
  }
}
```
Here's a little test YAML that cycles through all the characters in the first digit position:

```
display:
# Custom component for Quad-Alphanumeric LED display by Samuel Sieb
# See https://github.com/ssieb/custom_components/tree/master/lcd_ht16k33
  - platform: lcd_ht16k33
    address: 0x70
    update_interval: 500ms
    id: led_display
    lambda: |-
      it.set_brightness(0.5);
      it.printf("%c", id(counter));
      if (++id(counter) > 127)
        id(counter) = 0;
```

ssieb, 2020-03-07:
Ok, I'll swap those bytes.
I was thinking about the decimal point afterwards and considering options of how to add it, but then I forgot about it.  I agree that lookahead would work.  I'll add that.

ssieb, 2020-03-07:
I've pushed the changes, let me know if I got them right.
I used float for the brightness to match the usual usage in esphome.  It hides the implementation detail so that you don't have to know how many levels it actually has.

patfelst, 2020-03-08:
almost there! Needs the following changes.

For decimal point:
https://github.com/ssieb/custom_components/blob/084d518c24da1577406ead26835a0b5f92ebaec0/lcd_ht16k33/ht16k33_display.cpp#L74 should be
`fontc |= 0x4000;`

While the code structure is easier to understand now, the byte order is still reversed, I fixed by swapping these lines
https://github.com/ssieb/custom_components/blob/084d518c24da1577406ead26835a0b5f92ebaec0/lcd_ht16k33/ht16k33_display.cpp#L77-L78 to this:
```
    this->buffer_[pos++] = fontc & 0xff;
    this->buffer_[pos++] = fontc >> 8;
```
Finally the buffer sizes for printf and strftime both need to be increased from 4 to 6 to allow for decimal point and string termination characters. This is at lines 87 and 96, i.e.
`char buffer[6];`

I tested the above changes with print, printf and strftime (format `"%H.%M"`, i.e colons won't work). I also tested too many characters `it.printf("%u", 1234567);` and it handles this well displaying "1234" and the log says "writing off the screen!". I tested decimal point in the 4th digit using `it.print("1.2.3.4.");` (all digits actually 😀). All of this is working OK.

ssieb, 2020-03-08:
Ok, should be fixed now.  I made the buffers long enough for a dot on every character *and* the terminator.  It will have to be changed when scrolling is implemented, but it will work for now.

patfelst, 2020-03-08:
All fixed now, this issue can be closed - thanks!

closed 2020-03-08 by patfelst