ssieb/esphome_components #3: Feature request: scrolling text for HT16K33 4 character alphanumeric

patfelst, 2020-03-08:
Once we have fixed the basics for the HT16K33 display as per [this issue](https://github.com/ssieb/custom_components/issues/2#issuecomment-596153809)

This might be pushing the friendship, but what are the chances of implementing scrolling text for this?

I think about 20 characters would be nice, just simple scrolling from right to left and then repeat? I'd be happy to code it if you give me a starting point with the method structure! Not sure if there should be a parameter in the YAML `scroll: true` or simply make it automatic when buffer length > 4 chars? Might need a speed parameter too like `scroll_speed: 250ms` which is the pause time for each character.

Let me know your thoughts - thanks!

ssieb, 2020-03-08:
I've submitted a change to allow up to 32 characters of scrolling.

patfelst, 2020-03-08:
It's working very well first time for scrolling a single line of text, if that's all you want to do, it is done.

But for alternating between multiple lines of text it needs a bit more work. It took me a while to understand what's happening under the hood with respect to the buffer and the scroll, which is important when you're displaying multiple lines of text. The behaviour seems to be that there is a single buffer of text which gets scrolled and the state of the scroll remains the same even if a new line of text is written to it. I think the state of the scroll (it might be the dwell time) should be reset whenever a new line of text is written.

For example with the following YAML:

```
display:
  - platform: lcd_ht16k33
    address: 0x70
    update_interval: 1s
    scroll: true
    scroll_speed: 100ms
    scroll_dwell: 3s
    id: led_display
    lambda: |-
      it.set_brightness(0.4);

      auto time = id(homeassistant_time).now();
      if ((time.second % 15) < 10)
        it.printf("Temperature %.1f%c", id(temperature).state, 27);
      else
        it.strftime("Time %H.%M.%S", id(homeassistant_time).now());
```
"Let's say temperature xx.x°" has just been scrolled a few times. After 10 seconds, the time of day is written to the display, however you don't see "Time HH." scroll by first, you just see "MM.SS" because it's being shown on the previous scroll state for temperature as the dwell time hasn't expired. If the timing is right, after the dwell time finishes, you then see the entire time string scroll.

Hope that makes sense.

ssieb, 2020-03-08:
Yes, I understand and had considered this.  The problem is that your update_interval is 1s.  If the scroll  resets on every update, you would never even get to the end of the buffer, never mind waiting at the end.  I could keep 2 buffers and compare them and only reset if it changes.  But then, if the time changes during the scroll, you wouldn't want that to reset the scroll either, just update the time in the current buffer.  I suppose it could trigger on the length changing, you would just have to make sure there was a length difference between the different messages.

patfelst, 2020-03-09:
ahh yes of course the 1 second updates, I should have thought of that! I think leave it as-is for now, maybe just add a bit of an explanation to the component instructions. As long as the end user understands how it works, they should be able to adjust their yaml accordingly.

Do you have a "buy me a coffee" link? I'd like to thank you for all the work you've done on this and the CWWW2 custom components! Actually I was thinking of sending you one of these displays as a thank you, if this of interest, send me your details to patrickfelstead@gmail.com. Cheers!

patfelst, 2020-03-09:
> maybe just add a bit of an explanation to the component instructions.

I suggest something like "Note that the timing of each scroll update is not synchronised with changes to the text being displayed. So if you are displaying multiple lines of text, keep the scroll time (set by scroll_speed) + dwell time at least half of the update period of each line of text. This will ensure that the entire line of text will be scrolled at least twice before the text is changed to the next line. e.g. for a dwell time of 3s, make sure each line of text is displayed for at least 6s."

patfelst, 2020-03-09:
One other thing, there almost needs to be a bit of dwell time at the start of the string as the first character disappears pretty quickly. I got around this by putting in two spaces at the beginning of all my print statements - this works quite well.

patfelst, 2020-03-09:
I wasn't going to open a new issue for this but I added some comments to the font.h file:

```
static const uint16_t alphafonttable[] PROGMEM =  {
0b0000000000000000, // 0 = space
0b0000000000000001, // 1 = seg A
0b0000000000000010, // 2 = seg B
0b0000000000000100, // 3 = seg C
0b0000000000001000, // 4 = seg D
0b0000000000010000, // 5 = seg E
0b0000000000100000, // 6 = seg F
0b0000000001000000, // 7 = seg G1
0b0000000010000000, // 8 = seg G2
0b0000000100000000, // 9 = seg H
0b0000001000000000, // 10 = seg I
0b0000010000000000, // 11 = seg J
0b0000100000000000, // 12 = seg K
0b0001000000000000, // 13 = seg L
0b0010000000000000, // 14 = seg M
0b0100000000000000, // 15 = seg N (Decimal Point)
0b1000000000000000, // 16 = (not used, for offset to char 32)
0b0000000000000000, // 17 = (not used, for offset to char 32)
0b0000000000000000, // 18 = (not used, for offset to char 32)
0b0000000000000000, // 19 = (not used, for offset to char 32)
0b0000000000000000, // 20 = (not used, for offset to char 32)
0b0000000000000000, // 21 = (not used, for offset to char 32)
0b0000000000000000, // 22 = (not used, for offset to char 32)
0b0000000000000000, // 23 = (not used, for offset to char 32)
0b0001001011001001, // 24 = € (Euro?)
0b0001010111000000, // 25 = ¥ (Yen)
0b0001001011111001, // 26 = ? (unknown)
0b0000000011100011, // 27 = ° (degrees)
0b0000010100110000, // 28 = ? (unknown)
0b0001001011001000, // 29 = ± (plus-or-minus)
0b0011101000000000, // 30 = ↑ (up arrow)
0b0001011100000000, // 31 = ↓ (down arrow)
0b0000000000000000, // 32 = space
```

ssieb, 2020-03-09:
I've added the scroll reset when the buffer length changes and a start delay option.

patfelst, 2020-03-09:
works beautifully, really nice, thank you

patfelst, 2020-03-10:
One more thing to note now that I've been testing a bit more. When a new different length string resets the scroll, it can happen mid way though the previous string before it reaches the end. If this is expected behaviour, I can live with it, at least you always see the start of the new string. -thanks

ssieb, 2021-01-24:
I think now it will not restart the scroll if the text changes.
I also added the comments, thank you.

closed 2021-01-24 by ssieb

patfelst, 2021-01-24:
ok thanks, working on other projects so I won't get to test this for some time.

ssieb, 2021-01-24:
I made that change a very long time ago, so you probably already have it.  I had left the issue open because I hadn't looked at adding the comments yet.