ssieb/esphome_components #4: Failed config,   Platform not found: 'display.lcd_ht16k33'.

DelroyTaylor, 2020-03-09:
Hello

After running esphome new.yaml run I receive the following error. I did comment out the working Adafruit code and replace with your upgrade.  

*** My ERROR >

Failed config

display.lcd_ht16k33: [source new4.yaml:171]

  Platform not found: 'display.lcd_ht16k33'.
  platform: lcd_ht16k33
  address: 112
  scroll: True
  scroll_speed: 250ms
  scroll_dwell: 2s
  lambda: it.print("ABCD");

*** from Yaml>

esphome:
  name: ht16k33led
  platform: ESP32
  board: nodemcu-32s
  arduino_version: latest
  includes: # get from github link below and put into /config/esphome
    # https://learn.adafruit.com/adafruit-led-backpack/0-dot-56-seven-segment-backpack
    # https://learn.adafruit.com/adafruit-led-backpack/downloads
    # https://github.com/adafruit/Adafruit_LED_Backpack
    
    # Removed Adafruit and my_text_sensor below for ssieb new i2c custom component 
    # https://github.com/ssieb/custom_components
    #- Adafruit_LEDBackpack.h
    #- Adafruit_LEDBackpack.cpp
    #- my_text_sensor.h
    - config\custom_components\lcd_ht16k33\ht16k33_display.h
    - config\custom_components\lcd_ht16k33\ht16k33_display.cpp
  libraries:
    #- "Adafruit GFX Library"
    #- "SPI"  # unused dependency

...

i2c:
  id: bus_a
  sda: 21
  scl: 22
  scan: False

display:
  - platform: lcd_ht16k33
    address: 0x70
    scroll: true
    scroll_speed: 250ms
    scroll_dwell: 2s
    lambda: |-
      it.print("ABCD");

ssieb, 2020-03-09:
It's not an include.  Please refer to the README on the front page. Feel free to suggest improvements to that.  You need the entire `lcd_ht16k33` directory from this repo in your `/config/esphome/custom_components` directory.

DelroyTaylor, 2020-03-09:
Thank you, Samuel, for the fast response. 
Your README instructions do not include "esphome" in the path. I however still have the "Failed config" error. Basically, where can/should the config folder be located? Should the config folder be in the folder containing the yaml file or the python folder?

ssieb, 2020-03-09:
I was assuming that you're using hassio. Either way, the custom_components directory has to be in the same directory as your yaml file.

DelroyTaylor, 2020-03-09:
I am not using hassio. I am using a Windows 8 laptop and i use OTA to upload. The project yaml file is in C:\Users\Delroy\custom. The custom component is locate here C:\Users\Delroy\custom\config\esphome\custom_components\lcd_ht16k33

ssieb, 2020-03-09:
`C:\Users\Delroy\custom\custom_components\lcd_ht16k33` would be the correct location.

DelroyTaylor, 2020-03-11:
Yes. This works. Thank you Samuel. Please update your README non hassio users.

closed 2020-03-11 by DelroyTaylor