ssieb/esphome_components #5: Feature Request: Method to send temperature to EC sensor.

mrgoose2, 2020-07-13:
The EC sensor [https://atlas-scientific.com/circuits/ezo-conductivity-circuit/](url) allows calibrating the sensor with the temperature to get more accurate readings. Their example code here : [https://github.com/Atlas-Scientific/Ezo_I2c_lib/tree/master/Examples/temp_comp_example](url)

Would it be possible for this to automatically take the temp reading from the RTD sensor if you have one? If not, how would you recommend periodically updating this temperature calibration?

ssieb, 2020-07-13:
I could add a method to allow you to send a temperature value to the sensor.

mrgoose2, 2020-07-13:
You are seriously amazing.

ssieb, 2020-07-13:
I see this is common function over several sensors so that is good.  Can I rely on the firmware being V2.13 or newer?

mrgoose2, 2020-07-13:
If you are referring to the stamps firmware, mine are not. 
EC,1.8
pH,1.9
RTD,1.2

Let me look into how to update them.

ssieb, 2020-07-13:
You probably should anyway, but that means I won't rely on the new feature.

mrgoose2, 2020-07-13:
I have no problem updating. I just need to figure out how.

ssieb, 2020-07-13:
I just pushed this.  It doesn't require the firmware update.

mrgoose2, 2020-07-13:
I'm having trouble getting this to work.

Here's my YAML

```
i2c:
  sda: D2
  scl: D1
  scan: True
  id: bus_a

  #ezo platform from: https://github.com/ssieb/custom_components
sensor:
  - platform: ezo
    filters:
      - lambda: return id(ph_ezo).set_tempcomp_value(rtd_ezo);
    id: ph_ezo
    name: "pH"
    address: 99
    accuracy_decimals: 2
    unit_of_measurement: "pH"
  
  - platform: ezo
    id: rtd_ezo
    name: "RTD Temperature"
    address: 101
    accuracy_decimals: 2
    unit_of_measurement: "°C"

  - platform: ezo
    id: ec_ezo
    name: "EC CN"
    address: 100
    accuracy_decimals: 0
    unit_of_measurement: "CN"
```

```

Show Logs aquaman.yaml
INFO Reading configuration /config/esphome/aquaman.yaml...
INFO Starting log output from aquaman.local using esphome API
INFO Connecting to aquaman.local:6053 (192.168.1.111)
INFO Successfully connected to aquaman.local
[15:45:06][I][app:100]: ESPHome version 1.14.5 compiled on Jul 13 2020, 15:34:16
[15:45:06][C][wifi:415]: WiFi:
[15:45:06][C][wifi:283]:   SSID: [redacted]
[15:45:06][C][wifi:284]:   IP Address: 192.168.1.111
[15:45:06][C][wifi:286]:   BSSID: [redacted]
[15:45:06][C][wifi:287]:   Hostname: 'aquaman'
[15:45:06][C][wifi:291]:   Signal strength: -65 dB ▂▄▆█
[15:45:06][C][wifi:295]:   Channel: 1
[15:45:06][C][wifi:296]:   Subnet: 255.255.255.0
[15:45:06][C][wifi:297]:   Gateway: 192.168.1.1
[15:45:06][C][wifi:298]:   DNS1: 192.168.1.35
[15:45:06][C][wifi:299]:   DNS2: (IP unset)
[15:45:06][E][ezo.sensor:062]: device returned no data
[15:45:06][E][ezo.sensor:062]: device returned no data
[15:45:06][E][ezo.sensor:062]: device returned no data
[15:45:06][C][i2c:028]: I2C Bus:
[15:45:06][C][i2c:029]:   SDA Pin: GPIO4
[15:45:06][C][i2c:030]:   SCL Pin: GPIO5
[15:45:06][C][i2c:031]:   Frequency: 50000 Hz
[15:45:06][I][i2c:033]: Scanning i2c bus for active devices...
[15:45:06][I][i2c:040]: Found i2c device at address 0x63
[15:45:06][I][i2c:040]: Found i2c device at address 0x64
[15:45:06][I][i2c:040]: Found i2c device at address 0x65
[15:45:06][E][ezo.sensor:062]: device returned no data
[15:45:06][E][ezo.sensor:062]: device returned no data
[15:45:06][E][ezo.sensor:062]: device returned no data
[15:45:06][C][logger:175]: Logger:
[15:45:06][C][logger:176]:   Level: DEBUG
[15:45:06][C][logger:177]:   Log Baud Rate: 115200
[15:45:06][C][logger:178]:   Hardware UART: UART0
[15:45:06][E][ezo.sensor:062]: device returned no data
[15:45:06][E][ezo.sensor:062]: device returned no data
[15:45:06][E][ezo.sensor:062]: device returned no data
[15:45:06][C][ezo.sensor:014]: EZO 'pH'
[15:45:06][C][ezo.sensor:014]:   Unit of Measurement: 'pH'
[15:45:06][C][ezo.sensor:014]:   Accuracy Decimals: 2
[15:45:06][C][ezo.sensor:015]:   Address: 0x63
[15:45:06][C][ezo.sensor:018]:   Update Interval: 60.0s
[15:45:06][E][ezo.sensor:062]: device returned no data
[15:45:06][E][ezo.sensor:062]: device returned no data
[15:45:06][E][ezo.sensor:062]: device returned no data
[15:45:06][C][ezo.sensor:014]: EZO 'RTD Temperature'
[15:45:06][C][ezo.sensor:014]:   Unit of Measurement: '°C'
[15:45:06][C][ezo.sensor:014]:   Accuracy Decimals: 2
[15:45:06][C][ezo.sensor:015]:   Address: 0x65
[15:45:06][C][ezo.sensor:018]:   Update Interval: 60.0s
[15:45:06][E][ezo.sensor:062]: device returned no data
[15:45:06][E][ezo.sensor:062]: device returned no data
[15:45:06][E][ezo.sensor:062]: device returned no data
[15:45:06][C][ezo.sensor:014]: EZO 'EC CN'
[15:45:06][C][ezo.sensor:014]:   Unit of Measurement: 'CN'
[15:45:06][C][ezo.sensor:014]:   Accuracy Decimals: 0
[15:45:06][C][ezo.sensor:015]:   Address: 0x64
[15:45:06][C][ezo.sensor:018]:   Update Interval: 60.0s
[15:45:06][E][ezo.sensor:062]: device returned no data
[15:45:06][E][ezo.sensor:062]: device returned no data
[15:45:06][E][ezo.sensor:062]: device returned no data
[15:45:06][C][ota:029]: Over-The-Air Updates:
[15:45:06][C][ota:030]:   Address: aquaman.local:8266
[15:45:06][C][ota:032]:   Using Password.
[15:45:06][E][ezo.sensor:062]: device returned no data
[15:45:06][E][ezo.sensor:062]: device returned no data
[15:45:06][E][ezo.sensor:062]: device returned no data
[15:45:06][C][api:095]: API Server:
[15:45:06][C][api:096]:   Address: aquaman.local:6053
[15:45:06][E][ezo.sensor:062]: device returned no data
[15:45:06][E][ezo.sensor:062]: device returned no data
[15:45:06][E][ezo.sensor:062]: device returned no data
[15:45:06][E][ezo.sensor:062]: device returned no data
[15:45:06][E][ezo.sensor:062]: device returned no data
[15:45:06][E][ezo.sensor:062]: device returned no data
[15:45:06][E][ezo.sensor:062]: device returned no data
[15:45:06][E][ezo.sensor:062]: device returned no data
[15:45:06][E][ezo.sensor:062]: device returned no data
```

The device returned no data message just continues.

ssieb, 2020-07-13:
It's not a filter. How often do you want to recalibrate it? You could either use an interval or watch for a certain amount of change in the temperature sensor.

mrgoose2, 2020-07-13:
I'm not very knowledgeable with these sensors or esp programming. Could you help me fix my yaml so it updates just periodically? I dont really have a set interval in mind, just some kind of temperature calibration.

ssieb, 2020-07-13:
```yaml
interval:
  interval: 30min
  then:
    - lambda: id(ph_ezo).set_tempcomp_value(rtd_ezo.state);
    - lambda: id(ec_ezo).set_tempcomp_value(rtd_ezo.state);
```

mrgoose2, 2020-07-14:
I think that worked. I dont really know how to validate that. I also still have the device returned no data message spamming the logs even with the logger debug level commented out.

ssieb, 2020-07-14:
Right, that shouldn't be happening.  I'll check.  I don't have one these sensors, so I have no way of testing it myself.

ssieb, 2020-07-14:
Ok, I found it.  It should be fixed now.

mrgoose2, 2020-07-14:
That seems to have done it! You are amazing! On the update interval. I did notice it doesn't actually implement the temperature correction until that first interval. Meaning you'll read incorrect (or using the default of 25C) for the first 30 minutes after boot until the lambda runs. I changed it to 1 minute and it was all set. Thank you so much for working with me!

ssieb, 2020-07-14:
That means you're adjusting the calibration for every reading, but I suppose that should be ok.  You could also add an on_boot script to wait for the first temperature measurement and then send it.

closed 2020-07-14 by ssieb