ssieb/esphome_components #8: Growatt - not expected output

sanderlv, 2020-12-10:
Have tried https://github.com/ssieb/custom_components/tree/master/growatt

This esphome:
```
esphome:
  name: growatt_modbus
  platform: ESP8266
  board: d1_mini

substitutions:
  esp_name: growatt_modbus

wifi:
  ssid: "IOTDOM"
  password: !secret IOTDOM_wifi_key
  use_address: 192.168.6.42
  domain: .local.local
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${esp_name} fb Hotspot"
    password: !secret Fallback_Hotspot

captive_portal:

# Enable logging
logger:
  level: VERBOSE
#  baud_rate: 9600

# Enable Home Assistant API
api:

ota:

time:
  - platform: homeassistant
    id: homeassistant_time

uart:
  tx_pin: D1
  rx_pin: D2
  baud_rate: 9600
  # stop_bits: 2

sensor:
  - platform: wifi_signal
    name: "${esp_name} - WiFi Signal"
    update_interval: 6s
    
  - platform: uptime
    name: "${esp_name} - Uptime"
    icon: mdi:clock-outline
    update_interval: 60s
    
  - platform: growatt
    id: my_growatt
    input_power:
      id: input_power
      name: "${esp_name} Input Power"
    pv1_voltage:
      id: pv1_voltage
      name: "${esp_name} PV1 Voltage"
    pv1_current:
      id: pv1_current
      name: "${esp_name} PV1 Current"
    pv1_power:
      id: pv1_power
      name: "${esp_name} PV1 Power"
    pv2_voltage:
      id: pv2_voltage
      name: "${esp_name} PV2 Voltage"
    pv2_current:
      id: pv2_current
      name: "${esp_name} PV2 Current"
    pv2_power:
      id: pv2_power
      name: "${esp_name} PV2 Power"
    grid_frequency:
      id: grid_frequency
      name: "${esp_name} Grid Frequency"
    output_power:
      id: output_power
      name: "${esp_name} Output Power"
    temperature:
      id: temperature
      name: "${esp_name} Temperature"

# https://github.com/ssieb/custom_components/tree/master/growatt
```

Gives this output:
[image]

sanderlv, 2020-12-10:
I do have a "Growatt Inverter Modbus RTU Protocol_II V1_20-English.pdf" which I am not allowed to publicly share.

ssieb, 2020-12-11:
Where did you get it from?
Does it look like http://www.photovoltaicsolar.in/Growatt_Manual/MAX%20Series%20Modbus%20RTU%20Protocol.pdf which appears to be version 1.05 instead of the 1.20 one you have?  The important part is Section 4.2 Input Reg.  Do the registers match?
That manual is actually quite a bit different from the one I originally looked at, so I'll see if this one makes any difference.

ssieb, 2020-12-11:
I had a major bug in the decoding and I've updated it with the registers from the new doc. Get the new .cpp file and see how it works.

sanderlv, 2020-12-11:
I get "different" values now:
[image]

ssieb, 2020-12-11:
If you look in the log, there should be a raw output line.  Can you paste that here so I can see what the actual values are?
Also, please compare the input registers from the document I linked to the one you have.  Are they the same?

sanderlv, 2020-12-11:
> 
> 
> If you look in the log, there should be a raw output line. Can you paste that here so I can see what the actual values are?
> Also, please compare the input registers from the document I linked to the one you have. Are they the same?

Sorry, I do no know how  to do this, the section 4.2 is pages long and I do not know where to look at...

Also 2 additional questions...

I think modbus send every second data, but now it only comes ~30 secs?
And how can I more quickly updat ethe home assistant entities?
I tried     update_interval: 1s but tha tis not accepted by esphome

sanderlv, 2020-12-11:
Some more>

[09:51:34][D][sensor:092]: 'growatt_modbus Temperature': Sending state 0.00000 °C with 1 decimals of accuracy
[09:51:34][V][app:081]: A component took a long time in a loop() cycle (0.23 s).
[09:51:34][V][app:082]: Components should block for at most 20-30ms in loop().
[09:51:34][V][sensor:013]: 'growatt_modbus - Uptime': Received new state 47.823002
[09:51:34](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[09:51:35][V][sensor:013]: 'growatt_modbus - Uptime': Received new state 48.805000
[09:51:35][D][sensor:092]: 'growatt_modbus - Uptime': Sending state 48.80500 s with 0 decimals of accuracy
[09:51:36][V][sensor:013]: 'growatt_modbus - Uptime': Received new state 49.811001
[09:51:36][D][sensor:092]: 'growatt_modbus - Uptime': Sending state 49.81100 s with 0 decimals of accuracy
[09:51:37][V][sensor:013]: 'growatt_modbus - Uptime': Received new state 50.805000
[09:51:37][D][sensor:092]: 'growatt_modbus - Uptime': Sending state 50.80500 s with 0 decimals of accuracy
[09:51:38][V][sensor:013]: 'growatt_modbus - Uptime': Received new state 51.812000
[09:51:38][D][sensor:092]: 'growatt_modbus - Uptime': Sending state 51.81200 s with 0 decimals of accuracy
[09:51:39][V][sensor:013]: 'growatt_modbus - WiFi Signal': Received new state -58.000000
[09:51:39][D][sensor:092]: 'growatt_modbus - WiFi Signal': Sending state -58.00000 dB with 0 decimals of accuracy
[09:51:39][D][growatt:017]: Data: 00.01.00.D9.00.00.00.64.00.00.27.10.00.00.C3.50.03.E8.41.4C.31.2E.30.00.5A.41.42.41.00.01.00.01.00.00.03.E8.00.3C.00.3C.00.5A.00.5A.00.00.44.47.46.30.41.30.39.30.31.41.00.00.00.00.00.01.00.64.00.00.00.00.20.20.20.50.56.20.49.6E.76.65.72.74.65.72.20.20 (84)
[09:51:39][D][growatt:055]: DATA: IP=1422131.2W, V1=10.0V, I1=0.000A, P1=65536000.0W, V2=5000.0V, I2=100.000A, P2=109551232.0W, OP=54213580.0W, F=188.0 Hz, TEMP=0.00
[09:51:39][V][sensor:013]: 'growatt_modbus Input Power': Received new state 1422131.250000
[09:51:39][D][sensor:092]: 'growatt_modbus Input Power': Sending state 1422131.25000 W with 1 decimals of accuracy
[09:51:39][V][sensor:013]: 'growatt_modbus PV1 Voltage': Received new state 10.000000
[09:51:39][D][sensor:092]: 'growatt_modbus PV1 Voltage': Sending state 10.00000 V with 1 decimals of accuracy
[09:51:39][V][sensor:013]: 'growatt_modbus PV1 Current': Received new state 0.000000
[09:51:39][D][sensor:092]: 'growatt_modbus PV1 Current': Sending state 0.00000 A with 3 decimals of accuracy
[09:51:39][V][sensor:013]: 'growatt_modbus PV1 Power': Received new state 65536000.000000
[09:51:39][D][sensor:092]: 'growatt_modbus PV1 Power': Sending state 65536000.00000 W with 1 decimals of accuracy
[09:51:39][V][sensor:013]: 'growatt_modbus PV2 Voltage': Received new state 5000.000000
[09:51:39][D][sensor:092]: 'growatt_modbus PV2 Voltage': Sending state 5000.00000 V with 1 decimals of accuracy
[09:51:39][V][sensor:013]: 'growatt_modbus PV2 Current': Received new state 100.000000
[09:51:39][D][sensor:092]: 'growatt_modbus PV2 Current': Sending state 100.00000 A with 3 decimals of accuracy
[09:51:39][V][sensor:013]: 'growatt_modbus PV2 Power': Received new state 109551232.000000
[09:51:39][D][sensor:092]: 'growatt_modbus PV2 Power': Sending state 109551232.00000 W with 1 decimals of accuracy
[09:51:39][V][sensor:013]: 'growatt_modbus Output Power': Received new state 54213580.000000
[09:51:39][D][sensor:092]: 'growatt_modbus Output Power': Sending state 54213580.00000 W with 1 decimals of accuracy
[09:51:39][V][sensor:013]: 'growatt_modbus Grid Frequency': Received new state 187.979996
[09:51:39][D][sensor:092]: 'growatt_modbus Grid Frequency': Sending state 187.98000 hz with 1 decimals of accuracy
[09:51:39][V][sensor:013]: 'growatt_modbus Output Power': Received new state 54213580.000000
[09:51:39][D][sensor:092]: 'growatt_modbus Output Power': Sending state 54213580.00000 W with 1 decimals of accuracy
[09:51:39][V][sensor:013]: 'growatt_modbus Temperature': Received new state 0.000000
[09:51:39](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[09:51:39][V][app:081]: A component took a long time in a loop() cycle (0.26 s).
[09:51:39][V][app:082]: Components should block for at most 20-30ms in loop().
[09:51:39][V][sensor:013]: 'growatt_modbus - Uptime': Received new state 52.828999
[09:51:39][D][sensor:092]: 'growatt_modbus - Uptime': Sending state 52.82900 s with 0 decimals of accuracy
[09:51:40][V][sensor:013]: 'growatt_modbus - Uptime': Received new state 53.804001
[09:51:40][D][sensor:092]: 'growatt_modbus - Uptime': Sending state 53.80400 s with 0 decimals of accuracy
[09:51:41][V][sensor:013]: 'growatt_modbus - Uptime': Received new state 54.810001
[09:51:41][D][sensor:092]: 'growatt_modbus - Uptime': Sending state 54.81000 s with 0 decimals of accuracy
[09:51:42][V][sensor:013]: 'growatt_modbus - Uptime': Received new state 55.805000
[09:51:42][D][sensor:092]: 'growatt_modbus - Uptime': Sending state 55.80500 s with 0 decimals of accuracy
[09:51:43][V][sensor:013]: 'growatt_modbus - Uptime': Received new state 56.810001
[09:51:43][D][sensor:092]: 'growatt_modbus - Uptime': Sending state 56.81000 s with 0 decimals of accuracy
[09:51:44][D][growatt:017]: Data: 00.01.00.D9.00.00.00.64.00.00.27.10.00.00.C3.50.03.E8.41.4C.31.2E.30.00.5A.41.42.41.00.01.00.01.00.00.03.E8.00.3C.00.3C.00.5A.00.5A.00.00.44.47.46.30.41.30.39.30.31.41.00.00.00.00.00.01.00.64.00.00.00.00.20.20.20.50.56.20.49.6E.76.65.72.74.65.72.20.20 (84)
[... cut ...]
[09:51:49][D][sensor:092]: 'growatt_modbus Input Power': Sending state 1422131.25000 W with 1 decimals of accuracy
[09:51:49][V][sensor:013]: 'growatt_modbus PV1 Voltage': Received new state 10.000000
[09:51:49][D][sensor:092]: 'growatt_modbus PV1 Voltage': Sending state 10.00000 V with 1 decimals of accuracy
[09:51:49][V][sensor:013]: 'growatt_modbus PV1 Current': Received new state 0.000000
[09:51:49][D][sensor:092]: 'growatt_modbus PV1 Current': Sending state 0.00000 A with 3 decimals of accuracy
[09:51:49][V][sensor:013]: 'growatt_modbus PV1 Power': Received new state 65536000.000000

ssieb, 2021-01-24:
Turned out I was using the wrong command.  This has been fixed and serveral more sensors added.

closed 2021-01-24 by ssieb