ssieb/esphome_components #11: A noob question: how to run a component?

Protoncek, 2021-04-07:
Hello!
I'm struggling, but with no success. Can someone please point me into correct procedure? I just don't know how to run "keypad" (or any other, to be honest) component. I copied keypad folder into config/custom_components, i included keypad_text_sensor.h into my ESPHome code, but i keep getting error : component "keypad" not found. 
What am i doing wrong? HA and ESPHome are wonderfull apps, it's just that beginner's help is way too hard to find...

ssieb, 2021-04-07:
You don't use the `includes:` for this.  It works just like the built-in components.  I think you've put the folder in the wrong place though.  You should have `/config/esphome/custom_components/keypad/__init__.py` for example.

DelroyTaylor, 2021-04-07:
"keypad" component? (#11)

ssieb, 2021-04-07:
@DelroyTaylor it looks like you replied to the email with no content, so I'm going to delete it.  If you meant to write something, then try again.

Protoncek, 2021-04-07:
Great, many thanks! Keypad now works! Now i have to send finished code into HA- how should i create text sensor to send finished code into HA? 
And: when i stop typing code and i DON'T press # - is is any  reset time or program holds partly entered code indefinitely?

(sorry for "dumb" questions, i'm pretty noob in these things... i'm still heavily learning...)

ssieb, 2021-04-07:
The keypad `text_sensor` already does the sending to HA.  It depends on what you want to do with the result.  There's also the `on_value` trigger of the text_sensor if you want to have other actions.
There is no reset timeout.  You could file an issue if you want that.

danyboy666, 2021-04-07:
Is there a way to use this with the PCF8574 componant? I guess I need a I2C keypad library or componant?

ssieb, 2021-04-07:
https://esphome.io/components/pcf8574.html
You can use the `pin:` definition there with the keypad.

ssieb, 2021-04-07:
I recommend coming to the esphome Discord server for more interactive help.

danyboy666, 2021-04-07:
I'll come check it out thank you.

Protoncek, 2021-04-08:
> The keypad `text_sensor` already does the sending to HA. It depends on what you want to do with the result. 

That's what i thought, too, but i get nothing in HA, nor i find any "keypad" sensor... i did a copy/paste of your example in my yaml.

> There is no reset timeout. You could file an issue if you want that.

It would be a very good idea to implement a reset after, say, 5-10 seconds. Currently it can happen a situation when someone  (passing by) accidentally (or intentionally) hits a number or two and when i come to enter a number it won't be correct one unless i hit asterix first an only then enter a number.

ssieb, 2022-11-20:
@belgianrubs just letting you know that there have been very large changes in how to use this component, so your documentation is out of date.  Did you write it before `external_components:` was available?  Also, I'm in the process of integrating the components to main esphome.

closed 2022-11-20 by ssieb

belgianrubs, 2023-01-04:
@ssieb Sorry for the late response, totally missed your question. I'm not sure about that, I downloaded the files back in the day and copied them into the custom_components folder. 
Great to hear that the component is moving to main ESPHome, I'm looking forward to see it pop-up there! Thanks for all the effort you're putting into ESPHome!