ssieb/esphome_components #13: cwww2 light does not compile in 2021.8.0

hydazz, 2021-08-19:
looks like in 2021.8.0 they changed color modes and its broken something

Problematic code: [here](https://github.com/ssieb/custom_components/blob/master/cwww2/cwww2_light_output.h#L19-L22)
```python
In file included from src/esphome.h:14:0,
                 from src/main.cpp:3:
src/esphome/components/cwww2/cwww2_light_output.h: In member function 'virtual esphome::light::LightTraits esphome::cwww2::CWWW2LightOutput::get_traits()':
src/esphome/components/cwww2/cwww2_light_output.h:19:12: error: 'class esphome::light::LightTraits' has no member named 'set_supports_brightness'
     traits.set_supports_brightness(true);
            ^
src/esphome/components/cwww2/cwww2_light_output.h:20:12: error: 'class esphome::light::LightTraits' has no member named 'set_supports_rgb'
     traits.set_supports_rgb(false);
            ^
src/esphome/components/cwww2/cwww2_light_output.h:21:12: error: 'class esphome::light::LightTraits' has no member named 'set_supports_rgb_white_value'
     traits.set_supports_rgb_white_value(false);
            ^
src/esphome/components/cwww2/cwww2_light_output.h:22:12: error: 'class esphome::light::LightTraits' has no member named 'set_supports_color_temperature'
     traits.set_supports_color_temperature(true);
            ^
*** [.pioenvs/arlec-globe/src/main.cpp.o] Error 1
```

ssieb, 2021-08-19:
I think you found out that there's a built-in component to handle this now.  I should remove this one from the repo.

closed 2021-08-19 by ssieb

botts7, 2021-09-30:
If anyone want a link its here https://esphome.io/components/light/color_temperature.html

Need to invert Output for color_temperature

```
output:
  - platform: esp8266_pwm
    id: output1
    pin: GPIO14
  - platform: esp8266_pwm
    id: output2
    pin: GPIO12
    inverted: True
    

light:
  - platform: color_temperature
    id: LED  
    name: $project
    color_temperature: output2
    brightness: output1
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2700 K
    default_transition_length: 2.0s    
    gamma_correct: 1.0
    # Ensure the light turns on by default if the physical switch is actuated.
    restore_mode: ALWAYS_ON
```