ssieb/esphome_components #18: Can't compile the ht16k33_alpha custom component

ShadyMCShadesen, 2021-12-25:
I am not able to compile using the **ht16k33_alpha** custom component.

### **Logs:**
[image]

### **Config:**

```yaml
globals:
  - id: on_boot
    type: bool
    restore_value: false
    initial_value: 'true'

substitutions:
  # Device informations
  device_friendly_name: "HT16K33 Test"
  device_name: "ht16k33-test"
  device_entity_id: "ht16k33_test"
  # Pinout
  i2c_a_sda_pin: GPIO21
  i2c_a_scl_pin: GPIO22
  
# Device configuration
esphome:
  name: $device_name
  platform: esp32
# board: heltec_wifi_kit_32_v2   ###   i have a board with 8 MB flash so i tried using this one without success
  board: esp32dev
  on_boot:
    priority: -100
    then:
      - lambda: |-
          id(on_boot) = false;

# Import custom_components
external_components:
  - source:
      type: local
      path: custom_components
    
# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret api_password
  encryption:
    key: !secret api_encryption_key

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: $device_friendly_name
    password: !secret fallback_password

captive_portal:

web_server: # Could be disabled if performance issues appear
  port: 80
  auth:
    username: !secret esphome_web_username
    password: !secret esphome_web_password

# Set up I2C bus
i2c:
  - id: bus_a
    sda: $i2c_a_sda_pin
    scl: $i2c_a_scl_pin
    scan: true

# Set up display component
display:
  - platform: ht16k33_alpha
    address: 0x71
    scroll: true
    scroll_speed: 250ms
    scroll_dwell: 2s
    scroll_delay: 3
    lambda: |-
      it.print("Ht16K33 Display Test");
    id: ${device_entity_id}_display
```

**Board: [Mouser.com](https://eu.mouser.com/ProductDetail/Espressif-Systems/ESP32-DevKitC-VE?qs=sGAEpiMZZMv0NwlthflBi%252Br5FanRtbmfx%252BNQlezsXmY%3D&countrycode=DE&currencycode=EUR)**

**Display: [Mouser.com](https://eu.mouser.com/ProductDetail/Seeed-Studio/104020133?qs=sGAEpiMZZMv0NwlthflBi4vO7rZgU5LOSNDIFep7ods%3D&countrycode=DE&currencycode=EUR)**

ShadyMCShadesen, 2021-12-30:
Less errors but still not compilable

[image]

ssieb, 2021-12-30:
Turns out it only happens for esp32.  I tested it only with esp8266.
Unrelated, but there's no benefit to using substitutions in an `id:`.

ShadyMCShadesen, 2021-12-30:
> Turns out it only happens for esp32. I tested it only with esp8266. 
Ok, i'll try using one.

> Unrelated, but there's no benefit to using substitutions in an `id:`.
It works so i use it to rename all my entity_id's in one go if i change my device name any time in the future.

ssieb, 2021-12-31:
I'll have the ESP32 fixed shortly.
The `id:` is not visible anywhere other than internal to the ESP whether or not it's an internal component.

closed 2021-12-31 by ssieb

ShadyMCShadesen, 2021-12-31:
Here we go again (sorry)

[image]

ssieb, 2021-12-31:
I saw that in the previous logs and I don't know why you're getting that.  Try cleaning your build directory first.

ssieb, 2021-12-31:
Also, you should add `components: [ ht16k33_alpha ]` to the `external_components:` section to make sure you aren't picking up anything else.

ShadyMCShadesen, 2021-12-31:
> I saw that in the previous logs and I don't know why you're getting that. Try cleaning your build directory first.

> Also, you should add `components: [ ht16k33_alpha ]` to the `external_components:` section to make sure you aren't picking up anything else.

Tried it, sadly still the same errors. Would it work if i'll add the namespace of "memset" in front of it? (Like 'namespace::memset') (I don't know the namespace of memset)

ssieb, 2021-12-31:
Which version of esphome do you have?  I just took your config from above and compiled it with no errors.  I am running dev, but it shouldn't be significantly different than the previous release.

ShadyMCShadesen, 2021-12-31:
`v2021.12.1`

Well, i tried it with the Dev Version and it works. Thanks (⊃｡•́‿•̀｡)⊃

ShadyMCShadesen, 2021-12-31:
I think the font is wrong for my module, the scrolling works and interacting with different brightness levels does too.

[imgur](https://i.imgur.com/qn6Fx13.mp4)

ssieb, 2021-12-31:
The scrolling doesn't look like it's working right there either.  The one on the far right isn't changing.  You could open another issue about this if you want.  I'll see if I can get one of these modules, but it will be a while.

commit 87375dda (2021-12-27): Fix ht16k33_alpha compile errors
Fixes #18
commit 6934456a (2021-12-31): Fix compile errors on ESP32
Fixes #18