ssieb/esphome_components #19: ht16k33_alpha: Rendering issues with module

ShadyMCShadesen, 2021-12-31:
_This issue is based on:_
> The scrolling doesn't look like it's working right there either.  The one on the far right isn't changing.  You could open another issue about this if you want.  I'll see if I can get one of these modules, but it will be a while.
>
> _Originally posted by @ssieb in https://github.com/ssieb/custom_components/issues/18#issuecomment-1003272942_

I have a rendering issue with my HT16K33 module. Interacting with for example brightness works fine, scrolling seems to work at least for the first 3 digits (of 4), but letters or numbers (and symbols i guess) are hard to recognize ;).

### Code
```yaml
globals:
  - id: on_boot
    type: bool
    restore_value: false
    initial_value: 'true'

substitutions:
  # Device informations
  device_friendly_name: "HT16K33 Test"
  device_name: "ht16k33-test"
  device_entity_id: "ht16k33_test"
  # Pinout
  i2c_a_sda_pin: GPIO21
  i2c_a_scl_pin: GPIO22

# Device configuration
esphome:
  name: $device_name
  platform: esp32
  board: heltec_wifi_kit_32_v2
  on_boot:
    priority: -100
    then:
      - lambda: |-
          id(on_boot) = false;
      - script.execute: down # start brightness-test-script

# Import custom_components
external_components:
  - source:
      type: local
      path: custom_components
    components: [ ht16k33_alpha ]
    
# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret api_password
  encryption:
    key: !secret api_encryption_key

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: $device_friendly_name
    password: !secret fallback_password

captive_portal:

web_server: # Could be disabled if performance issues appear
  port: 80
  auth:
    username: !secret esphome_web_username
    password: !secret esphome_web_password

# Test script to see if changing the brightness works
script:
  - id: up
    then:
      - lambda: "id(${device_entity_id}_display).set_brightness(0.8);"
      - delay: 5s
      - script.execute: down
  - id: down
    then:
      - lambda: "id(${device_entity_id}_display).set_brightness(0.2);"
      - delay: 5s
      - script.execute: up

# Set up I2C bus
i2c:
  - id: bus_a
    sda: $i2c_a_sda_pin
    scl: $i2c_a_scl_pin
    scan: true

# Set up display component
display:
  - platform: ht16k33_alpha
    address: 0x71
    scroll: true
    scroll_speed: 250ms
    scroll_dwell: 2s
    scroll_delay: 3
    lambda: |-
      it.print("HT16K33 Display Test");
    id: ${device_entity_id}_display
```

### Video
[**imgur**](https://i.imgur.com/qn6Fx13.mp4)

### Hardware
[**ESP32 (Mouser.com)**](https://eu.mouser.com/ProductDetail/Espressif-Systems/ESP32-DevKitC-VE?qs=sGAEpiMZZMv0NwlthflBi%252Br5FanRtbmfx%252BNQlezsXmY%3D)
[**HT16K33 (Mouser.com)**](https://eu.mouser.com/ProductDetail/Seeed-Studio/104020133?qs=sGAEpiMZZMv0NwlthflBi4vO7rZgU5LOSNDIFep7ods%3D)

ssieb, 2022-02-02:
I have a module now and I copied most of your config.  It works perfectly for me, so I don't know what is happening with yours.

ShadyMCShadesen, 2022-02-02:
> I have a module now and I copied most of your config.  It works perfectly for me, so I don't know what is happening with yours.

Yeah i've ordered some of the adafruit ht16k33 backpack modules and it works fine. I think my module i bought first is damaged.

closed 2022-02-02 by ssieb