ssieb/esphome_components #26: feature request: custom buffer size [ht16k33_alpha]

ShadyMCShadesen, 2022-04-23:
Hey,
it would be awesome if the ht16k33_alpha component supports a optional setting for the buffer size (default is 64, i had to display stuff that is long than this, so i increased it to 128 myself)

Something like this for example:
``` yaml
display:
  - platform: ht16k33_alpha
    address: 0x71
    buffer: 128 # <-- custom buffer setting (default to 64)
    scroll: true
    scroll_speed: 250ms
    scroll_dwell: 2s
    scroll_delay: 3
    lambda: |-
      it.print("HT16K33 Display Test");
    id: ${device_entity_id}_display
```

ssieb, 2022-04-24:
That's a rather long scroll for a 4 character display. :astonished:

ShadyMCShadesen, 2022-10-22:
https://github.com/ssieb/custom_components/pull/36

ssieb, 2024-01-13:
I have implemented unlimited buffer size.  Have fun!

closed 2024-01-13 by ssieb