ssieb/esphome_components #28: Feature Request: Desky  - Stop movement

OkhammahkO, 2022-04-25:
It would be handy if desky component users had a way to stop the desk when it is in motion.

In my tests I just used a button which turned off both the up and the down pins (which need to be inverted;).

Thank you!

```
button: 
#Stop movement 
  - platform: template
    name: Stop Desk
    id: desky_stop_desk
    on_press:
      then:
        - switch.turn_off: raise_desk
        - switch.turn_off: lower_desk
        

switch:
#wake up ther desk and request it sends its height 
  - platform: gpio
    id: wake_desk_and_get_height
    name: "Request Desk Height"
    pin:
      number: D5
      inverted: true
    on_turn_on:
    - delay: 100ms
    - switch.turn_off: wake_desk_and_get_height

#Raise the desk 
  - platform: gpio
    id: raise_desk
    name: "Raise Desk"
    pin:
      number: D2
      # mode: INPUT_PULLUP
      inverted: true
    interlock: lower_desk
    on_turn_on:
    #Auto off after 15s just in case
    - delay: 15s
    - switch.turn_off: raise_desk
#Lower the desk 
  - platform: gpio
    id: lower_desk
    name: "Lower Desk" 
    pin:
      number: D1
      # mode: INPUT_PULLUP
      inverted: true
    interlock: raise_desk
    on_turn_on:
   #Auto off after 15s just in case
    - delay: 15s
    - switch.turn_off: lower_desk    

```

ssieb, 2022-05-28:
I added a `.stop()` method and an optional timeout value.

OkhammahkO, 2022-05-28:
Thank you!  I'm still chipping away at the project and will take a look over the next week...

closed 2022-05-28 by ssieb

OkhammahkO, 2022-05-30:
Tested and confirmed working! Thank you!

commit 524dd84c (2022-05-28): add some Desky features
fixes #27, fixes #28, fixes #29, fixes #30