ssieb/esphome_components #31: combining ezo and i2c_button functionality

Assunto, 2022-05-05:
Hi!  I'm trying to find a way to control the Atlas Scientific EZO Pumps  (I believe you've worked on the sensors before) in ESPHome because there isn't a way to turn on or off an i2c sensor and that's what the EZO devices are classified as.

I found your i2c_button component and believe it would work for the pumps if there was the ability to turn them on/off virtually (like a switch component) instead of reading the state of the physical button presses (like a sensor). I added the i2c_button component to my esphome config file and assigned one of my pumps to it. When I flashed the firmware the pump was recognized and spammed the "Button B Long Clicked" log.

I'm sure it's just a tiny bit of code to make a hybrid "i2c_switch" component but unfortunately my coding skills are terrible.

ssieb, 2022-05-18:
The i2c_button component is for reading from a button.  It's not for controlling a switch.
This EZO pump has a lot of controlling options, it's not a simple on/off switch.  I'll see what I can come up with.

Assunto, 2022-05-18:
Thank you **so** much!  I feel like I've scoured every component in esphome and none of them appear to give control to an i2c device without requiring a gpio pin too.

I'm not sure how difficult or time consuming it is to add all the controlling options but if it's easier or quicker, all I really need is the ability to id the pump in the .yaml and code it to turn on and off. From there I can figure out how much volume is dispensed by the second and create an accurate automation from there.

Again, thank you so so much

Assunto, 2022-06-16:
It looks like everything is happening via a number of pull requests (#3528, #3561, #2133, #2115, and #1179) and the work of @gvdhoven and @carlos-sarmiento

Thank you very much for helping us all out to get these additions implemented

closed 2022-06-16 by Assunto