ssieb/esphome_components #32: matrix_keypad not working since rename

DrBanarne, 2022-09-18:
I had the keypad component working before the rename to matrix_keypad. Now esphome editor says "Component not found: keypad". The same goes for binary_sensor.keypad. I have changed components section in my yaml from keypad to matrix_keypad. If I exclude the components part, esphome editor says "Component keypad cannot be loaded via YAML (no CONFIG_SCHEMA)."
I have tried to load the components (matrix_keypad and key_provider) locally (matrix_keypad folder renamed to keypad) but then esphome compiler can't find the file "key_provider.h".
What might be wrong, am I missing something?

ssieb, 2022-09-18:
Sorry, I thought I had tested it after the rename.  The binary_sensor was still requiring `keypad`.  I just pushed a fix.

r0667002, 2022-09-26:
It's still not working for me

ssieb, 2022-09-26:
What's happening?

r0667002, 2022-09-26:
**ESPHome version:** 2022.9.1

I'm using the following code:

```
esphome:
  name: alarm-control
  platform: ESP8266
  board: d1_mini

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: !secret ap_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Fallback Hotspot"
    password: !secret ap_password

captive_portal:

external_components:
  - source: github://ssieb/custom_components

matrix_keypad:
  id: mykeypad
  rows:
    - pin: D0
    - pin: RX
    - pin: TX
    - pin: D3
  columns:
    - pin: D4
    - pin: D5
    - pin: D6
    - pin: D7
  keys: "123A456B789C*0#D"
  has_diodes: false

binary_sensor:
  - platform: matrix_keypad
    keypad_id: mykeypad
    name: keyA
    id: keyA
    row: 0
    col: 3
  - platform: matrix_keypad
    keypad_id: mykeypad
    name: keyB
    id: keyB
    row: 1
    col: 3
  - platform: matrix_keypad
    keypad_id: mykeypad
    name: keyC
    id: keyC
    row: 2
    col: 3
  - platform: matrix_keypad
    keypad_id: mykeypad
    name: keyD
    id: keyD
    row: 3
    col: 3

text_sensor:
  - platform: matrix_keypad
    id: reader
    name: insertedcode
    keypad_id: mykeypad
    max_length: 4
    end_keys: "#"
    back_keys: "*"
    allowed_keys: "0123456789"
    on_progress:
      - logger.log: 
          format: "input progress: '%s'"
          args: [ 'x.c_str()' ]
    on_value:
      - logger.log: 
          format: "input result: '%s'"
          args: [ 'x.c_str()' ]
```

But getting the error messages:

- Component binary_sensor.matrix_keypad requires component keypad.
- Platform not found: 'text_sensor.matrix_keypad'.

If I just try `keypad` as a component instead of `matrix_keypad`, I get the no CONFIG_SCHEMA error as listed above.

DrBanarne, 2022-09-26:
I tried it again now since it up til now still complained about that it could not find component keypad and matrix_keypad didn't work either. 
Today matrix_keypad does not generate any errors in the editor but as r0667002 says I get error: "Component binary_sensor.matrix_keypad requires component keypad."
If I remove the binary_sensor section it validates, but when it compiles I get error: 

In file included from src/esphome/components/matrix_keypad/keypad.cpp:1:
src/esphome/components/matrix_keypad/keypad.h:3:10: fatal error: esphome/components/key_provider/key_provider.h: No such file or directory
    3 | #include "esphome/components/key_provider/key_provider.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

ssieb, 2022-10-22:
Oops!  I committed it locally, but forgot to push it!  :man_facepalming: 
For the second error, did you include `key_provider` in the external component config?

DrBanarne, 2022-10-24:
Yes, it's included. Tried with "components: [ matrix_keypad, key_provider ]" and also without to be sure. I still get this when installing:

Processing light-controller-house (board: d1_mini; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- ESPAsyncTCP-esphome @ 1.2.3
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- ESPAsyncTCP-esphome @ 1.2.3
|   |-- Hash @ 1.0
|   |-- ESP8266WiFi @ 1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- Wire @ 1.0
Compiling /data/light-controller-house/.pioenvs/light-controller-house/src/esphome/components/matrix_keypad/keypad.cpp.o
In file included from src/esphome/components/matrix_keypad/keypad.cpp:1:
src/esphome/components/matrix_keypad/keypad.h:3:10: fatal error: esphome/components/key_provider/key_provider.h: No such file or directory
    3 | #include "esphome/components/key_provider/key_provider.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Compiling /data/light-controller-house/.pioenvs/light-controller-house/src/esphome/components/mdns/mdns_esp8266.cpp.o
Compiling /data/light-controller-house/.pioenvs/light-controller-house/src/esphome/components/mdns/mdns_esp_idf.cpp.o
*** [/data/light-controller-house/.pioenvs/light-controller-house/src/esphome/components/matrix_keypad/keypad.cpp.o] Error 1
Compiling /data/light-controller-house/.pioenvs/light-controller-house/src/esphome/components/network/util.cpp.o
========================= [FAILED] Took 12.21 seconds =========================

r0667002, 2022-10-24:
Works for me! Except for the text sensor. Any reason you removed the text sensor?

Working code for others (downloaded the folders manually and put them in `esphome/custom_components`

```
esphome:
  name: alarm-control
  platform: ESP8266
  board: d1_mini

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: !secret ap_password

wifi:
  ssid: !secret wifi_ssid_upstairs
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Fallback Hotspot"
    password: !secret ap_password

captive_portal:

external_components:
  - source:
      type: local
      path: custom_components

matrix_keypad:
  id: mykeypad
  rows:
    - pin: D0
    - pin: RX
    - pin: TX
    - pin: D3
  columns:
    - pin: D4
    - pin: D5
    - pin: D6
    - pin: D7
  keys: "123A456B789C*0#D"
  has_diodes: false

binary_sensor:
  - platform: matrix_keypad
    keypad_id: mykeypad
    name: keyA
    id: keyA
    row: 0
    col: 3
  - platform: matrix_keypad
    keypad_id: mykeypad
    name: keyB
    id: keyB
    row: 1
    col: 3
  - platform: matrix_keypad
    keypad_id: mykeypad
    name: keyC
    id: keyC
    row: 2
    col: 3
  - platform: matrix_keypad
    keypad_id: mykeypad
    name: keyD
    id: keyD
    row: 3
    col: 3
```

ssieb, 2022-10-24:
> Works for me! Except for the text sensor. Any reason you removed it?

Yes.  I added the `input_builder` instead which can handle keys from other components (specifically wiegand for now) as well.

ssieb, 2022-10-24:
> Working code for others (downloaded the folders manually and put them in `esphome/custom_components`

That's not necessary and will stop you from getting any future updates.

DrBanarne, 2022-10-25:
Thanx for the fixes. Everything is now compiling correct and working as before. Keep up the good work! I will give the input_builder a go now.

closed 2022-10-25 by ssieb