ssieb/esphome_components #33: Feature request: support raw hex Wiegand values or non-standard bit lengths

jsb5151, 2022-10-03:
I have multiple cards that do not follow the 26-bit or 34-bit standard format, for example HID H10314 (40-bit) as well as other high-frequency tags that are either 56 or 64-bit.  I'd like to get the raw hex value into either the text sensor or as a tag value; this would let me use special credentials on my HID readers. 

BTW, thanks for the awesome work on this! :-)

staticlama, 2023-01-11:
This would be great. Some of my tags scan correctly, others have an error of unknown 36 bit value. Unfortunately the fobs are enrolled in another access system so I can't easily change them at this stage.

ssieb, 2023-01-12:
This has been added.  Use `url: https://github.com/ssieb/custom_components@wiegand` for the external component.
https://github.com/ssieb/custom_components/tree/wiegand/components/wiegand
This component is in process of getting merged to esphome.

closed 2023-01-12 by ssieb