ssieb/esphome_components #35: Feature-Request: Continuous scrolling (ht16k33)

ShadyMCShadesen, 2022-10-15:
Hey, would it be possible to have a setting for continuous scrolling of the ht16k33_alpha display component. 🙂

Currently the scroll stops at the end of the text and jumps back to the start. The "continuous"-version would act a little different:
If the scrolling text comes to an end, the beginning is appended with some distance and stops briefly as soon as it has arrived at the beginning again. then the scroll continues and everything starts from the beginning

Probably with timing settings for the scroll and the pause

😊

ShadyMCShadesen, 2022-10-22:
https://github.com/ssieb/custom_components/pull/36

ssieb, 2024-01-13:
I have implemented this now, please test it out.

closed 2024-01-13 by ssieb