ssieb/esphome_components #42: Wiegand : delete TAG after transmit

stmb-100, 2022-11-23:
Wiegand:
Is there a way to delete the tag of the RFID reader after transmission or to set it to 'xxxx' so that the tag is not accidentally transmitted again.

Gibt es eine Möglichkeit den Tag des RFID Readers nach der Übertragung zu löschen oder auf 'xxxx' zu setzen, damit der Tag nicht ungewollt erneut übertragen wird.

ssieb, 2022-11-24:
It doesn't work like that any more.  There's no text_sensor and no stored value.  Just the `on_tag:` trigger which is triggered once per scan.

stmb-100, 2022-11-24:
Hi
thanks that sounds good, then I'll try it once and integrate it into IOBroker. Has it already been tested with IOBroker?

danke das hört sich gut an,  dann werde ich es einmal versuchen und in IOBroker einbinden. Wurde es schon mit IOBroker getestet?

stmb-100, 2022-11-24:
Hi
In the log I see the read RFID TAG. But it doesn't arrive in IOBroker ESPHOME Device. In the MQTT device I only see the message in the debug. How do I get the TAG in the ESPHomeDevice

ESPHOME Log
[17:25:53][V][wiegand.text_sensor:048]: received 26-bit value: 22efc4e
[17:25:53][D][wiegand.text_sensor:051]: received 26-bit tag: 1539623
[17:25:53][D][TEST:145]: received tag 1539623

MQTT Device debug
[0;36m[D][TEST:145]: received tag 1539623 [0m

ssieb, 2022-11-24:
I don't know what IOBroker is or how it works, but it sounds like it uses mqtt.  At the place where your config sends the `[17:25:53][D][TEST:145]: received tag 1539623` log message now, you can change that to send an mqtt message instead.  Or use a template text_sensor and publish to that.

stmb-100, 2022-11-24:
I tried it like this and it works for now

text_sensor:   
  - platform: template
    name: "RFID Tag"
    id: rfid_tag

wiegand:
  - id: reader
    d0: D1
    d1: D2
    on_tag:
      - lambda: ESP_LOGD("TEST", "received tag %s", x.c_str());
      - text_sensor.template.publish:
          id: rfid_tag
          state: !lambda 'return x;'   
    on_key:
      - lambda: ESP_LOGD("TEST", "received key %d", x);

stmb-100, 2022-11-25:
Unfortunately, an update of the TAG (text_sensor) is sent every 60 seconds. The update_interval of the text_sensor cannot be deactivated and therefore IOBrocker gets the TAG sent every 60 seconds

ssieb, 2022-11-25:
I didn't think it would be sending anything if it hasn't changed.  But you can disable it, `update_interval: never`.  You should be using a direct mqtt publish anyway instead of a text_sensor.

closed 2023-02-22 by ssieb