ssieb/esphome_components #45: ip5306

MichaKersloot, 2023-01-03:
Hi,

I'm trying to use the ip5306 component, but it doesn't seem to work?

**Config**:

i2c:
  - id: bus_a
    sda: 14
    scl: 13
    frequency: 400kHz
    scan: true    

ip5306:
  battery_level:   # sensor
    name: Battery Level
  charger_connected:  # binary_sensor
    id: connected
    on_press:
      then:
        - lambda: ESP_LOGD("TEST", "charging");
    on_release:
      then:
        - lambda: ESP_LOGD("TEST", "not charging");
  charge_full:  # binary_sensor
    id: full
    on_press:
      then:
        - lambda: ESP_LOGD("TEST", "fully charged");
    on_release:
      then:
        - lambda: ESP_LOGD("TEST", "still charging");
 
**Logging:**
[11:02:09][C][i2c.arduino:052]: I2C Bus:
[11:02:09][C][i2c.arduino:053]:   SDA Pin: GPIO14
[11:02:09][C][i2c.arduino:054]:   SCL Pin: GPIO13
[11:02:09][C][i2c.arduino:055]:   Frequency: 400000 Hz
[11:02:09][C][i2c.arduino:058]:   Recovery: bus successfully recovered
[11:02:09][I][i2c.arduino:068]: Results from i2c bus scan:
[11:02:09][I][i2c.arduino:074]: Found i2c device at address 0x3C
[11:02:09][I][i2c.arduino:074]: Found i2c device at address 0x75

But nothing from ip5306, while I expected something from:
`ESP_LOGD(TAG, "Setting up ip5306...");`

I'm using esphome with homeassistant and the 'Battery level' entity is available, but at 0%

ssieb, 2023-01-03:
In order to see setup messages, you have to watch the serial logs over USB.

MichaKersloot, 2023-01-06:
Ah, yes, that works. I also found out the battery level only updates on a device reset.

MichaKersloot, 2023-01-12:
As the battery level only updated on a device reset, I've changed the component from Component to PollingComponent. The diffs:

ip5306.h
```
@@ -8,10 +8,11 @@
 namespace esphome {
 namespace ip5306 {
 
-class IP5306 : public i2c::I2CDevice, public Component {
+class IP5306 : public i2c::I2CDevice, public PollingComponent {
  public:
+  IP5306() : PollingComponent(60000) {}
   void setup() override;
-  void loop() override;
+  void update() override;
 
   float get_setup_priority() const override;
```

ip5306.cpp
```
@@ -25,7 +25,7 @@ void IP5306::setup() {
   }
 }
 
-void IP5306::loop() {
+void IP5306::update() {
   uint8_t data[2];
   if (this->battery_level_ != nullptr) {
     if (this->read_register(IP5306_REG_LEVEL, data, 1) != i2c::ERROR_OK) {
@@ -40,8 +40,7 @@ void IP5306::loop() {
       case 0x80: value = 75; break;
       case 0x00: value = 100; break;
     }
-    if (!this->battery_level_->has_state() || (this->battery_level_->state != value))
-      this->battery_level_->publish_state(value);
+  this->battery_level_->publish_state(value);
   }
   if (this->read_register(IP5306_REG_READ0, data, 2) != i2c::ERROR_OK) {
     ESP_LOGE(TAG, "unable to read status");
```

landonr, 2023-01-16:
what device are you using? I think my m5stack fire gets battery level updates with the current implementation but I'll check to be sure

MichaKersloot, 2023-01-16:
Hi, I've activated the ip5306 on my T-Journal as stated in https://discord.com/channels/429907082951524364/951213660905287761/1061049798402265228 . It looks like the first 
the removed if statement is not doing what you expect? I suspect it only publishes the state on the first run to create it and never runs  again. To me it also seems better to use the PollingComponent instead of Component to limit the updates to a reasonable rate.

ssieb, 2024-01-11:
I don't see what that PR fixes.  The current code will update when the battery level changes.  If that's not working, then polling is not going to work either since it's still the same code.

MichaKersloot, 2024-01-11:
The code only publishes on a device reset. Or maybe when the battery level changes, i've not seen this happening. This means when Homeassistant restarts there is no value reported and it's shown as NA. So changing it to a PollinComponent that reports the current value on polling fixes this. Both changes work together.