ssieb/esphome_components #47: VBUS: Questions and Warning regarding Resol Deltasol CS Plus

Gloomyeye, 2023-01-09:
Hello,

I installed the VBUS controller for my Resol Deltasol CS Plus regarding to this feature-request: https://github.com/esphome/feature-requests/issues/1949. So far, the new device/config is working. But I have the following questions:

I get the following warning in my home assistant log:

```
Logger: homeassistant.components.sensor
Source: components/sensor/__init__.py:980
Integration: Sensor (documentation, issues)
First occurred: 8. Januar 2023 um 14:49:47 (1 occurrences)
Last logged: 8. Januar 2023 um 14:49:47

Entity sensor.hwr_sensor_deltasol_sc_r1_flow_rate (<class 'homeassistant.components.esphome.sensor.EsphomeSensor'>) is using native unit of measurement 'L/h' which is not a valid unit for the device class ('water') it is using; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+esphome%22
```
It is correct that the [water device class](https://developers.home-assistant.io/docs/core/entity/sensor/#available-device-classes) is used for the flow_rate sensor? Theoretically, it should be a device class like "water_speed" (this is not given).

The error sensors will not be exposed to Home Assistant. Is my config for model "Deltasol CS Plus" correct? In ESPHome is sensors are defined.

In addition, the sensor "heat_quantity" shows only a 0 value. Is that normal or there should be value? My installation is the same as "Anlage 1: Standard-Solaranlage" of the [manual](https://www.resol.de/Produktdokumente/11202699_DeltaSol_CS_Plus.monde.pdf) from page 10.

This is my config:
```
external_components:
  - source:
      type: git
      url: https://github.com/ssieb/custom_components
      ref: vbus
    components: [ vbus ]
    refresh: 1min

uart:
  id: resol
  rx_pin: GPIO20
  baud_rate: 9600

vbus:
  uart_id: resol

sensor:
  - platform: vbus
    model: deltasol_cs_plus
    temperature_1:
      id: hwr_sensor_deltasol_sc_kol_temperature #KOL
      name: "${friendly_name} Kollektor Temperatur"
      device_class: "temperature"
      state_class: measurement
      unit_of_measurement: "°C"
      accuracy_decimals: 1
    temperature_2:
      id: hwr_sensor_deltasol_sc_tsp_temperature #TSP
      name: "${friendly_name} Speicher Temperatur"
      device_class: "temperature"
      state_class: measurement
      unit_of_measurement: "°C"
      accuracy_decimals: 1
    flow_rate:
      id: hwr_sensor_deltasol_sc_r1_flow_rate
      name: "${friendly_name} Relais Drehzahl" #R1
      #unit_of_measurement: "l/h"
    heat_quantity:
      id: hwr_sensor_deltasol_sc_heat_quantity
      name: "${friendly_name} Wärmemenge" #Heat Quantity
      state_class: measurement
      #unit_of_measurement: "Wh"
    operating_hours_1:
      id: hwr_sensor_deltasol_sc_r1_operating_hours
      name: "${friendly_name} Relais Betriebsstunden" #Operating Hours
      device_class: "duration"
      unit_of_measurement: "h"
    pump_speed_1:
      id: hwr_sensor_deltasol_sc_pump_speed
      name: "${friendly_name} Pumpgeschwindigkeit" #Pump Speed 1
      #unit_of_measurement: "%"
    time:
      id: hwr_sensor_deltasol_sc_device_time
      name: "${friendly_name} Gerätezeit" #Device Time
      entity_category: diagnostic
      disabled_by_default: true

binary_sensor:
  - platform: vbus
    model: deltasol_cs_plus
    sensor1_error:
      id: hwr_sensor_deltasol_sc_sensor1_error
      name: "${friendly_name} Sensor 1 Fehler" #Sensor 1 Error
      entity_category: diagnostic
    sensor2_error:
      id: hwr_sensor_deltasol_sc_sensor2_error
      name: "${friendly_name} Sensor 2 Fehler" #Sensor 2 Error
      entity_category: diagnostic
```
Thank you!

ssieb, 2023-01-09:
You're adding a lot of extra parameters to the sensors.  Most of that is already defined in the component.
If you don't like that warning from HA about the unit, then configure a different unit.  You could file an issue on HA to add that class and units.
The binary sensor won't show up by default because they are diagnostic category.  You have to add them.

Gloomyeye, 2023-01-10:
I checked the two __init__.py files of the vbus component and reduced my config with no extra information. This is my new config:

```
sensor:
  - platform: vbus
    model: deltasol_cs_plus
    temperature_1:
      id: hwr_sensor_deltasol_sc_kol_temperature #KOL
      name: "${friendly_name} Kollektor Temperatur"
    temperature_2:
      id: hwr_sensor_deltasol_sc_tsp_temperature #TSP
      name: "${friendly_name} Speicher Temperatur"
    flow_rate:
      id: hwr_sensor_deltasol_sc_r1_flow_rate
      name: "${friendly_name} Relais Drehzahl" #R1
      device_class: ""
      icon: mdi:cog
    heat_quantity:
      id: hwr_sensor_deltasol_sc_heat_quantity
      name: "${friendly_name} Wärmemenge" #Heat Quantity
    operating_hours_1:
      id: hwr_sensor_deltasol_sc_r1_operating_hours
      name: "${friendly_name} Relais Betriebsstunden" #Operating Hours
    pump_speed_1:
      id: hwr_sensor_deltasol_sc_pump_speed
      name: "${friendly_name} Pumpgeschwindigkeit" #Pump Speed 1
    time:
      id: hwr_sensor_deltasol_sc_device_time
      name: "${friendly_name} Gerätezeit" #Device Time
      disabled_by_default: true
    #version:
    #  id: hwr_sensor_deltasol_sc_device_firmware_version
    #  name: "${friendly_name} Geräte Firmware Version" #Device firmware version
    #  icon: "mdi:package"

binary_sensor:
  - platform: vbus
    model: deltasol_cs_plus
    sensor1_error:
      id: hwr_sensor_deltasol_sc_sensor1_error
      name: "${friendly_name} Sensor 1 Fehler" #Sensor 1 Error
    sensor2_error:
      id: hwr_sensor_deltasol_sc_sensor2_error
      name: "${friendly_name} Sensor 2 Fehler" #Sensor 2 Error
```

I added an icon and an empty device class to the flow rate sensor. Currently, the water device class for the flow rate sensor is wrong. Maybe you can change this in the coding.

I found a copy&paste failure in line 234 [binary_sensor](https://github.com/ssieb/custom_components/tree/vbus/components/vbus/binary_sensor)/__init__.py . This should be "CONF_DELTASOL_CS_PLUS". Therefore, no error sensors will be created in my config. I can not add them manually. They do not show up in home assistant.

What do you think?

ssieb, 2023-01-10:
What do you think the class for the flow rate should be?

I fixed the binary sensor.

Gloomyeye, 2023-01-10:
Thanks for fixing. Now, the error sensors for my model show up in home assistant :-)

From my point of view, there should not be added a device class (or empty device class) to the flow rate sensor because so far no device class is suitable: https://developers.home-assistant.io/docs/core/entity/sensor/#available-device-classes.
A new device class for L/h needs to be added to home assistant. It should be like wind_speed.

ssieb, 2023-01-10:
ok, it's removed.

closed 2023-01-10 by ssieb