ssieb/esphome_components #50: Invalid parity

IgorSimic, 2023-02-19:
Since commit e48ea30 [add 37-bit and refactor parity checking] i have error after tag reading:
[09:16:37][I][RAW:131]: received raw 34 bits, value 200a9dfea
[09:16:37][D][wiegand.text_sensor:080]: received 34-bit tag: 5566453
[09:16:38][W][wiegand.text_sensor:082]: invalid parity

Same error for 34 and 26 bit readers.
I didn't change anything in my hardware.
I traced problem to the function;
`bool check_oparity(uint64_t value, int start, int length) {
  int parity = 0;
  uint64_t mask = 1LL << start;
  for (int i = 0; i <= length; i++, mask <<= 1)
    if (value & i)
      parity++;
  return parity & 1;
}`

If I change last line to return !(parity & 1); or return 1, everiyhing works just fine.

ssieb, 2023-02-22:
This is fixed now.  But also, this component has been merged to esphome, so you should use the built-in component now.  This fix will be in the next release.

closed 2023-02-22 by ssieb

IgorSimic, 2023-02-22:
Thank you very much!