ssieb/esphome_components #53: Debugging mode via uart_mitm does not work

DivanX10, 2023-05-16:
Could you add the ability to work in debug mode to uart_mitm?

I use the uart_mitm component for the Philips EP 5444 Coffee machine, so that I can control the coffee machine from the control panel and at the same time download the data for decryption. There is a ready-made project [ESPHome-Philips-Smart-Coffee](https://github.com/TillFleisch/ESPHome-Philips-Smart-Coffee) and it is suitable for coffee machines of the 1200-2200 series, and for the 5400 series it is necessary to find bits for control.

```
external_components:
  - source:
      type: git
      url: https://github.com/ssieb/custom_components
    components: [ uart_mitm ]

uart_mitm:
  uart1: uart_display
  uart2: uart_mainboard

uart:
#UART connection to the control panel. We read the data from the control panel and send it to the motherboard
 - rx_pin: GPIO16 #Reading data from the control panel
   tx_pin: GPIO17 #We send the data to the control panel
   baud_rate: 115200
   id: uart_display
   stop_bits: 1
   data_bits: 8
   parity: none
   debug:
     direction: TX
     dummy_receiver: true

#UART connection to the motherboard. We read the data from the motherboard and send it to the control panel
 - rx_pin: GPIO3 #Reading data from the motherboard
   tx_pin: GPIO1 #We send data to the motherboard
   baud_rate: 115200
   id: uart_mainboard
   stop_bits: 1
   data_bits: 8
   parity: none
   debug:
     direction: TX
     dummy_receiver: true
```

I can only use a parallel connection for debagging, and then I don't need to use this uart_mitm component. See the diagram in the figure
[image]

If I need to embed the ESP into the wiring and for the ESP to receive and send data, then I can't do without the uart_mitm component, but it works if I don't enable debugging mode. If I turn on debugging mode, the screen goes off and the coffee machine is not controlled from the control panel
[image]

ssieb, 2023-05-16:
`dummy_receiver: true` steals the bytes, so the component doesn't see them.  You only need that line if there's no component listening.

closed 2023-05-16 by ssieb

ssieb, 2023-05-16:
That would only be true if you didn't have the `mitm` component reading from the uart.