ssieb/esphome_components #60: Esmart3

Mtkodat, 2023-10-13:
Bonjour,
Je n'arrive pas à compiler sous ESPHOME ver: 2023.9.3,  dans Home Assistant.

J'ai essayé beaucoup de chose mais j'en arrive à être obligé de demander votre aide.
Je vous donne mon "yaml" et le log.

[esp8266-mppt-esmart3-40a.yaml.txt](https://github.com/ssieb/custom_components/files/12893605/esp8266-mppt-esmart3-40a.yaml.txt)

[logs_esp8266-mppt-esmart3-40a_run (1).txt](https://github.com/ssieb/custom_components/files/12893563/logs_esp8266-mppt-esmart3-40a_run.1.txt)

ssieb, 2023-10-13:
If you had tried that `includes:` method earlier, try doing a clean build.

Mtkodat, 2023-10-13:
J'ai 55 ans
J'ai juste besoin d'aide.
Je ne comprends pas votre réponse.
Des gens comme moi il y en a beaucoup. On cherche une aide concrète, visuelle, un exemple pour les non-développeurs.
Merci qd même.

ssieb, 2023-10-13:
[image]
Click on the entry that says "Clean Build Files".

Mtkodat, 2023-10-13:
Je le fais à chaque fois. Mais toujours pareil.

INFO ESPHome 2023.9.3
INFO Reading configuration /config/esphome/esp8266-mppt-esmart3-40a.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esp8266-mppt-esmart3-40a (board: nodemcuv2; framework: arduino; platform: ***@***.***)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/ESPAsyncTCP-esphome @ 2.0.0
INFO Installing esphome/ESPAsyncTCP-esphome @ 2.0.0
Unpacking  [####################################]  100%
Library Manager: ***@***.*** has been installed!
INFO ***@***.*** has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 3.1.0
INFO Installing esphome/ESPAsyncWebServer-esphome @ 3.1.0
Unpacking  [####################################]  100%
Library Manager: ***@***.*** has been installed!
INFO ***@***.*** has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing ottowinter/ESPAsyncTCP-esphome
INFO Installing ottowinter/ESPAsyncTCP-esphome
Unpacking  [####################################]  100%
Library Manager: ***@***.*** has been installed!
INFO ***@***.*** has been installed!
Library Manager: Installing esphome/noise-c @ 0.1.4
INFO Installing esphome/noise-c @ 0.1.4
Unpacking  [####################################]  100%
Library Manager: ***@***.*** has been installed!
INFO ***@***.*** has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.1
INFO Installing esphome/libsodium @ 1.10018.1
Unpacking  [####################################]  100%
Library Manager: ***@***.*** has been installed!
INFO ***@***.*** has been installed!
Library Manager: Installing bblanchon/ArduinoJson @ 6.18.5
INFO Installing bblanchon/ArduinoJson @ 6.18.5
Unpacking  [####################################]  100%
Library Manager: ***@***.*** has been installed!
INFO ***@***.*** has been installed!
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/esp8266-mppt-esmart3-40a/src/esmart3.cpp.o
Compiling .pioenvs/esp8266-mppt-esmart3-40a/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/esp8266-mppt-esmart3-40a/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/esp8266-mppt-esmart3-40a/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/esp8266-mppt-esmart3-40a/src/esphome/components/api/api_pb2_service.cpp.o
Compiling .pioenvs/esp8266-mppt-esmart3-40a/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/esp8266-mppt-esmart3-40a/src/esphome/components/api/list_entities.cpp.o
Compiling .pioenvs/esp8266-mppt-esmart3-40a/src/esphome/components/api/proto.cpp.o
Compiling .pioenvs/esp8266-mppt-esmart3-40a/src/esphome/components/api/subscribe_state.cpp.o
Compiling .pioenvs/esp8266-mppt-esmart3-40a/src/esphome/components/api/user_services.cpp.o
Compiling .pioenvs/esp8266-mppt-esmart3-40a/src/esphome/components/binary_sensor/automation.cpp.o
Compiling .pioenvs/esp8266-mppt-esmart3-40a/src/esphome/components/binary_sensor/binary_sensor.cpp.o
Compiling .pioenvs/esp8266-mppt-esmart3-40a/src/esphome/components/binary_sensor/filter.cpp.o
Compiling .pioenvs/esp8266-mppt-esmart3-40a/src/esphome/components/captive_portal/captive_portal.cpp.o
Compiling .pioenvs/esp8266-mppt-esmart3-40a/src/esphome/components/esmart3/esmart3.cpp.o
Compiling .pioenvs/esp8266-mppt-esmart3-40a/src/esphome/components/esp8266/core.cpp.o
Compiling .pioenvs/esp8266-mppt-esmart3-40a/src/esphome/components/esp8266/gpio.cpp.o
Compiling .pioenvs/esp8266-mppt-esmart3-40a/src/esphome/components/esp8266/preferences.cpp.o
Compiling .pioenvs/esp8266-mppt-esmart3-40a/src/esphome/components/homeassistant/time/homeassistant_time.cpp.o
[... 242 lines cut ...]
Compiling .pioenvs/esp8266-mppt-esmart3-40a/FrameworkArduino/stdlib_noniso.cpp.o
Compiling .pioenvs/esp8266-mppt-esmart3-40a/FrameworkArduino/time.cpp.o
Compiling .pioenvs/esp8266-mppt-esmart3-40a/FrameworkArduino/uart.cpp.o
Compiling .pioenvs/esp8266-mppt-esmart3-40a/FrameworkArduino/umm_malloc/umm_info.c.o
Compiling .pioenvs/esp8266-mppt-esmart3-40a/FrameworkArduino/umm_malloc/umm_integrity.c.o
Compiling .pioenvs/esp8266-mppt-esmart3-40a/FrameworkArduino/umm_malloc/umm_local.c.o
Compiling .pioenvs/esp8266-mppt-esmart3-40a/FrameworkArduino/umm_malloc/umm_malloc.cpp.o
Compiling .pioenvs/esp8266-mppt-esmart3-40a/FrameworkArduino/umm_malloc/umm_poison.c.o
Archiving .pioenvs/esp8266-mppt-esmart3-40a/libFrameworkArduino.a
Linking .pioenvs/esp8266-mppt-esmart3-40a/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/esp8266-mppt-esmart3-40a/src/esphome/components/esmart3/esmart3.cpp.o: in function `_ZN7esphome7esmart316ESmart3Component6updateEv':
esmart3.cpp:(.text._ZN7esphome7esmart316ESmart3Component6updateEv+0x10): multiple definition of `_ZN7esphome7esmart316ESmart3Component6updateEv'; .pioenvs/esp8266-mppt-esmart3-40a/src/esmart3.cpp.o:esmart3.cpp:(.text._ZN7esphome7esmart316ESmart3Component6updateEv+0x10): first defined here
[... cut ...] art316ESmart3Component4loopEv':
esmart3.cpp:(.text._ZN7esphome7esmart316ESmart3Component4loopEv+0x24): multiple definition of `_ZN7esphome7esmart316ESmart3Component4loopEv'; .pioenvs/esp8266-mppt-esmart3-40a/src/esmart3.cpp.o:esmart3.cpp:(.text._ZN7esphome7esmart316ESmart3Component4loopEv+0x24): first defined here
collect2: error: ld returned 1 exit status
*** [.pioenvs/esp8266-mppt-esmart3-40a/firmware.elf] Error 1
========================= [FAILED] Took 484.17 seconds =========================

ssieb, 2023-10-13:
> Compiling .pioenvs/esp8266-mppt-esmart3-40a/src/esmart3.cpp.o

Paste your current config.  That file should not be there, so you've done something.

Mtkodat, 2023-10-15:
Bonjour,
désolé je ne comprends pas votre question.
A savoir que j'ai recommencé avec un ESP32 avec le même fichier yaml et cela fonctionne.
Merci qd même.

ssieb, 2023-10-15:
That's because changing to the ESP32 or changing the name cleaned the build directory so it doesn't have that extra file now.

closed 2023-10-15 by ssieb

linalolo72, 2024-01-10:
@Mtkodat Bonjour, avez-vous réussi pour votre Esmart 3 ?

Mtkodat, 2024-01-10:
Bonjour
Oui

linalolo72, 2024-01-10:
ok super

patmtp35, 2024-02-25:
@mtkodat bonjour,

je suis en train de faire la même chose que vous même tranche d'age pas simple de se remettre au code .
je pense que pour ma part j'ai un souci de com avec le RS-485 vous avez cablé comment entre esmart est esp32 , j'utilise un max485 pour ma part

Mtkodat, 2024-02-25:
Bonjour,

Esmart3
1 | RS-485 A/+
2 | RS-485 B/-
5,6 | GND
7,8 | +5V out

Le 5v de l'esmart3 est suffisamment puissant pour alimenter vos deux modules.

patmtp35, 2024-02-25:
merci pour votre reponse rapide :)
ca c 'est bon j'ai via le git de skagmo .

mon j'ai cette connexion coté max485 
7,8 => vcc
A ==>A
B => B
5,6 => DND

de l 'autre coté vers l'esp32
RO=>RX GPIO1
DI=>TX GPIO3
DE,RE => GPIO22
 
je pense que c 'est la le pb car j'ai rien qui remonte

Mtkodat, 2024-02-25:
[image]
Avec ce modèle je n'ai pas eu a réfléchir

patmtp35, 2024-02-25:
ah super j'en ai un je crois dans un tirroir j'essaye merci !

patmtp35, 2024-02-25:
UN grand MErci cela fonctionne nickel !!

Mtkodat, 2024-02-25:
Parfait

patmtp35, 2024-03-01:
@Mtkodat 

afin de reduire le montage j'ai essayer avec une D1_mini. j'ai la led dur RS485 qui clignote qd le request part , mais j'ai aucun retour remonté d'info . tu avait le meme souci ?

spirake, 2024-12-25:
I want to implement a mppt esmart3 of 60A in home assistant with esp 8266, can we help with the yahal code?
Thank you

spirake, 2025-01-03:
please help, I don't receive data from esmart3 with the converter above. I don't see data in webserial either

spirake, 2025-01-03:
substitutions:
  device_name: esp8266-mppt-esmart3-60a
  adress_ip: "192.168.100.109"
  friendly_name: esmart3
  time_timezone: "Europe/Bucharest"
  
esphome:
  name: ${device_name}
  platform: ESP8266
  board: esp12e
#  includes:
#  - my_components/esmart3/esmart3.h
#  platformio_options:
#    upload_speed: 115200
#  includes:
#    - esmart3.h
#    - esmart3.cpp

external_components:
  - source:
      type: git
      url: https://github.com/ssieb/custom_components
    components: esmart3

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: ${adress_ip}
    gateway: 192.168.100.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp8266-Mppt-Esmart3-60A"
    password: "###########"

  reboot_timeout: 5min
  fast_connect: true

logger:
  # baud_rate: 0
  logs:
    component: ERROR

# Enable Home Assistant API
api:
  encryption:
    key: "#####################"

ota:
  - platform: esphome
    password: "###################f"

web_server:
  port: 80

interval:
  
  - interval: 60s
    then:
      if:
        condition:
          wifi.connected:
        else:
          - switch.turn_off: WIFI_LED
        then:
          - switch.turn_on: WIFI_LED
          
time:
  - platform: homeassistant
    timezone:  ${time_timezone} 

captive_portal:

uart:
    id: uart_bus
    tx_pin: GPIO1
    rx_pin: GPIO3
    baud_rate: 9600

#i2c:
#  sda: GPIO00 #
#  scl: GPIO02 #
#  scan: true
#  id: bus_a

#Etat de la connection
binary_sensor:
  - platform: status
    name: "${friendly_name} Status"

switch:    
  - platform: restart
    name: "${friendly_name} Restart"

  # Led połączenia wifi
  - platform: gpio
    pin: GPIO2
    inverted : True
    name: "WIFI_LED"
    id : WIFI_LED
    
text_sensor:
  # IP address #
  - platform: wifi_info
    ip_address:
      name: "${friendly_name} IP address"
      icon: "mdi:network-outline"
    ssid:
      name: "${friendly_name} Connected SSID"
    bssid:
      name: "${friendly_name} Connected BSSID"

  - platform: version
    id: text_sensor_version
    name: "${friendly_name} ESPHome Version"

sensor:
  - platform: wifi_signal
    name: "${friendly_name} WiFi Signal Sensor"
    update_interval: 30s

  - platform: uptime
    name: "${friendly_name} Uptime Sensor"
    update_interval: 30s

  - platform: esmart3
    charge_mode:
      id: charge_mode
      name: "${friendly_name} charge_mode"
    input_voltage:
      id: input_voltage
      name: "${friendly_name} input_voltage"
    battery_voltage:
      id: battery_voltage
      name: "${friendly_name} battery_voltage"
    charging_current:
      id: charging_current
      name: "${friendly_name} charging_current"
    load_voltage:
      id: load_voltage
      name: "${friendly_name} load_voltage"
    load_current:
      id: load_current
      name: "${friendly_name} load_current"
    charging_power:
      id: charging_power
      name: "${friendly_name} charging_power"
    load_power:
      id: load_power
      name: "${friendly_name} load_power"
    battery_temp:
      id: battery_temp
      name: "${friendly_name} battery_temp"
    internal_temp:
      id: internal_temp
      name: "${friendly_name} internal_temp"
    battery_level:
      id: battery_level
      name: "${friendly_name} battery_level"

linalolo72, 2025-01-03:
substitutions:
  updates: 5s
  name: esmart3
  temps: 10s
  decimales: "2"
  reboottime: 10min

esphome:
  name: gestion-esmart3
  friendly_name: gestion-esmart3
  
web_server:
  port: 80

esp8266:
  board: esp01_1m

# Enable logging
logger:
#  baud_rate: 0   # disable logging via UART, help to avoid numerous crash with ESP_LOGD
#  level: INFO  # INFO for less log, put DEBUG to view all the linky's received info in the logs
#  esp8266_store_log_strings_in_flash: False
  

# Enable Home Assistant API
api:
  encryption:
    key: "HWp0a2GXpCXjfALx2qS7fKYHwkRgV5BvxbNqJnNZ66o="

ota:
  - platform: esphome
    password: "dc9b15b049f8bfe2e7de08d59b099928"

wifi:
  networks:
    # - ssid: !secret wifi_ssid
    #   password: !secret wifi_password
    - ssid: !secret wifi_ssid1
      password: !secret wifi_password1 
    - ssid: !secret wifi_ssid2
      password: !secret wifi_password2
  reboot_timeout: ${reboottime}
  power_save_mode: NONE
  
  manual_ip:
    static_ip: 192.168.1.179
    gateway: 192.168.1.254
    subnet: 255.255.255.0 

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Gestion-Esmart3 Fallback Hotspot"
    password: "rzuTqFO5ssAj"

captive_portal:

# external_components:
#   - source:
#       type: git
#       url: https://github.com/ssieb/esphome_components
#     components: [ esmart3 ]
#     refresh: 0s

external_components:
  - source: components      
    components: [esmart3]
    refresh: 0s
    
uart:
  id: uartbus
[... 155 lines cut ...]
      }
################### retour info wifi de esp #################################
  - platform: wifi_info
    ip_address:
      name: ESP IP Address
      icon: mdi:ip
    ssid:
      name: ESP Connected SSID
      icon: mdi:box
    bssid:
      name: ESP Connected BSSID
    mac_address:
      name: ESP Mac Wifi Address
    scan_results:
      name: ESP Latest Scan Results
      icon: mdi:magnify-scan
    dns_address:
      name: ESP DNS Address 
      icon: mdi:dns
#################################################################################################################################

      
binary_sensor:
  - platform: status
    name: "${name}_status"

ssieb, 2025-01-03:
@spirake how did you wire it?
@linalolo72 your uart gpio numbers aren't valid.  Either `1` or `GPIO1`.

spirake, 2025-01-15:
Now I can't test because I'm not in the town, but I will test the first time I go

spirake, 2025-01-16:
Thank you linalolo72, I want to ask you why you have 2 wifi networks? Is the mppt connected to cloud box 1 or 2 to generate wifi? I don't understand. Thank you

linalolo72, 2025-01-19:
Hello, this will prevent any connection problems. Here is an esphome article => WiFi Component https://esphome.io/components/wifi.html

spirake, 2025-01-19:
`esphome:
  name: gestion-esmart3
  friendly_name: MPPT-esmart3
  platform: ESP8266
  board: d1_mini

# Enable logging
logger:
#  baud_rate: 0   # disable logging via UART, help to avoid numerous crash with ESP_LOGD
#  level: INFO  # INFO for less log, put DEBUG to view all the linky's received info in the logs
#  esp8266_store_log_strings_in_flash: False
  
substitutions:
  updates: 5s
  name: esmart3
  temps: 10s
  decimales: "2"
  reboottime: 10min
  
# Enable Home Assistant API
api:
  encryption:
    key: !secret api_encryption_key

ota:
  - platform: esphome
    password: !secret ota_password

wifi:
  networks:
    # - ssid: !secret wifi_ssid
    #   password: !secret wifi_password
    - ssid: !secret wifi_ssid1
      password: !secret wifi_password1 
    - ssid: !secret wifi_ssid2
      password: !secret wifi_password2
  reboot_timeout: ${reboottime}
  power_save_mode: NONE
  
  manual_ip:
    static_ip: 192.168.100.105
    gateway: 192.168.100.254
    subnet: 255.255.255.0 

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Gestion-Esmart3 Fallback Hotspot"
    password: !secret ap_password
    
web_server:
  port: 80

# external_components:
#   - source:
#       type: git
#       url: https://github.com/ssieb/esphome_components
#     components: [ esmart3 ]
#     refresh: 0s

external_components:
  - source:
      type: git
      url: https://github.com/ssieb/custom_components
    components: esmart3
    refresh: 0s
    
uart:
  id: uartbus
  tx_pin: GPIO4
  rx_pin: GPIO5
[... 162 lines cut ...]
  - platform: wifi_info
    ip_address:
      name: ESP IP Address
      icon: mdi:ip
    ssid:
      name: ESP Connected SSID
      icon: mdi:box
    bssid:
      name: ESP Connected BSSID
    mac_address:
      name: ESP Mac Wifi Address
    scan_results:
      name: ESP Latest Scan Results
      icon: mdi:magnify-scan
    dns_address:
      name: ESP DNS Address 
      icon: mdi:dns
#################################################################################################################################

      
binary_sensor:
  - platform: status
    name: "${name}_status"
          `
I have validated this code in HOA, can you see if it is correct? Did it work for you?

spirake, 2025-01-19:
I have validated this code in HOA, can you see if it is correct? Did it work for you?

linalolo72, 2025-01-19:
[Plan esphome8266 mini D1 ==>Esmart3 .pdf](https://github.com/user-attachments/files/18468926/Plan.esphome8266.mini.D1.Esmart3.pdf)

linalolo72, 2025-01-19:
[Plan esphome8266 mini D1 ==>Esmart3 .pdf](https://github.com/user-attachments/files/18468929/Plan.esphome8266.mini.D1.Esmart3.pdf)

linalolo72, 2025-01-19:
[image]

linalolo72, 2025-01-19:
[image]

linalolo72, 2025-01-21:
@spirake Hello, look above, there's my code which works fine

spirake, 2025-01-22:
Thank you, I made the wiring and installed the components, I also added a DHT22 for monitoring the temperature and humidity in the room.

[image]

[ESP8266+ESMART3+DHT22.pdf](https://github.com/user-attachments/files/18509202/ESP8266%2BESMART3%2BDHT22.pdf)

spirake, 2025-01-25:
linalolo72 how do you have the dashboard in the home assistant with the indicators on the esmart3, can you attach a picture? Thanks

linalolo72, 2025-02-13:
@spirake 

[image]