ssieb/esphome_components #64: Problem using magic_switch component

bernhardroth3, 2024-02-14:
Hi.

I try to use your magic_switch component in my HA/ESPHome installation.
During configuration all looks fine. On compilation however the linker complains about duplicate method definition as seen in the attached log.

Cant you help please.

Thank you.

[logs_magic-switch-01_run.txt](https://github.com/ssieb/esphome_components/files/14275404/logs_magic-switch-01_run.txt)
[magic-switch-01.yaml.txt](https://github.com/ssieb/esphome_components/files/14275425/magic-switch-01.yaml.txt)

ssieb, 2024-02-14:
Try doing a clean build.

bernhardroth3, 2024-02-14:
Thats in ESPHJome "Clean build files" then rebuild, right?  I have done that with the same results.

bernhardroth3, 2024-02-14:
For good measure I did it again and send the latest log attached.

[logs_magic-switch-01_run.txt](https://github.com/ssieb/esphome_components/files/14276496/logs_magic-switch-01_run.txt)

I use the latest versions of all subsystems:
ESPHome 2023.12.9

HA versions:
Core                      2024.2.1
Supervisor             2024.01.1
Operating System 11.5
Frontend                20240207.1

ssieb, 2024-02-15:
It looks like you have two copies somehow.  Try deleting that build directory or re-install the addon.

bernhardroth3, 2024-02-15:
OK. Found the problem. 
I indeed had multiple files inside the ESPHome docker container from earlier attempts to have all files local. 
Those attempts failed because the editor validator in ESPHome suddenly did not recognize the new magic-switch directive and I went back to referencing the Github repo. From the HA container it looked all was cleared but it was not in ESPHome. 

I certainly learned a bit about the inner workings of ESPHome and where to look if there is an issue. 
Nothing for the faint hearted though :-)

Thank you very much for your help and the great work you do for the HA and ESPHome community.

closed 2024-02-15 by ssieb