ssieb/esphome_components #66: Problem with magic_switch pulse timing report

snaulh, 2024-03-01:
Hi,
I would like to report a weird report
I set magic_switch with _timeout: 1200us_ but when checking log file from esphome, sometimes it did show like below

```

[20:32:48][D][magic_switch:019]: detected pulse of 28322us
[20:32:48][D][light:036]: 'Light' Setting:
[20:32:48][D][light:047]:   State: ON
[20:32:55][D][magic_switch:019]: detected pulse of 122710us
[20:32:55][D][light:036]: 'Light' Setting:
[20:32:55][D][light:047]:   State: OFF

```

Could you help to check it.
Thank you.

ssieb, 2024-03-01:
That's correct.  The `timeout:` setting is the minimum required to detect.  The log shows how long the pulse actually was.

closed 2024-03-01 by ssieb

snaulh, 2024-03-02:
Sorry and thank you. I got your point now,