ssieb/esphome_components #67: Could not find components folder for source.

SteveRMann, 2024-03-10:
I am trying to use an Adafruit HT16K33 with 14-segment LEDs.  When I put this at the top of my YAML code, I get the error in the title.

`external_components:
  - source:
      type: git
      url: https://github.com/ssieb/esphome_components
    components: [ ht16k33_alpha ]`

However, when I go to the URL, there is clearly a components folder.
What am I doing wrong?

Also, any chance of seeing a working yaml file using the 14-segment display?

ssieb, 2024-03-10:
So it is working?

SteveRMann, 2024-03-10:
Yes. Thanks for the quick reply. It took me a few minutes to figure out how to set the brightness, but that is also working now.

The only thing stopping me from moving my project forward is figuring out how to display the time.

ssieb, 2024-03-10:
It has the same `strftime` method as all the displays if you have a time component configured.  See the time docs for how to use it.

closed 2024-03-10 by ssieb