ssieb/esphome_components #69: Failed to compile the firmware with Magic Switch for Sonoff Basic R4

lnpbr, 2024-03-14:
**ESPHome version**
2024.2.2

**Project YAML**
esphome:
  name: sonoffbasicr4
  friendly_name: SonoffBasicR4
  platformio_options:
    board_build.mcu: esp32c3
    board_build.variant: esp32c3  

esp32:
  variant: ESP32C3
  board: esp32dev
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_BT_BLE_50_FEATURES_SUPPORTED: y
      CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y
      CONFIG_ESP_TASK_WDT_TIMEOUT_S: "10"

external_components:
  - source:
      type: git
      url: https://github.com/ssieb/esphome_components
    components: [ magic_switch ]

magic_switch:
  pin: 5
  on_switch:
      then:
        - light.toggle: light_1    

esp32_ble_tracker:
  scan_parameters:
    active: true

bluetooth_proxy:
  active: true

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO9
      mode:
        input: true
        pullup: true
      inverted: true
    id: button_1
    on_press:
      then:
        - light.toggle: light_1

output:
  - platform: gpio
    pin: GPIO4
    id: relay_1

light:
  - platform: binary
    name: Light
    id: light_1
    output: relay_1

status_led:
  pin:
    number: GPIO6
    inverted: yes

# Enable logging
logger:

**LOG**
Linking .pioenvs/sonoffbasicr4/firmware.elf
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: .pioenvs/sonoffbasicr4/src/main.o: in function `setup()':
/data/build/sonoffbasicr4/src/main.cpp:157: undefined reference to `vtable for esphome::magic_switch::MagicSwitch'
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /data/build/sonoffbasicr4/src/main.cpp:157: undefined reference to `vtable for esphome::magic_switch::MagicSwitch'
collect2: error: ld returned 1 exit status
*** [.pioenvs/sonoffbasicr4/firmware.elf] Error 1
========================= [FAILED] Took 14.74 seconds =========================

ssieb, 2024-03-15:
You're using esp-idf.  Do a clean build.

closed 2024-03-15 by ssieb

lnpbr, 2024-03-15:
It's working now, thanks 😁