ssieb/esphome_components #72: [pin] is an invalid option for [binary_sensor.seesaw]

lboue, 2024-04-08:
Hello,

I am facing an issue when I try to configure seesaw componant for my [Adafruit I2C Quad Rotary Encoder Breakout with NeoPixel - STEMMA QT / Qwiic, Product ID: 5752]( https://www.adafruit.com/product/5752):
`  [pin] is an invalid option for [binary_sensor.seesaw].`

These are the pin names in the seesaw firmware for each rotary encoder:
```
Encoder #0
  Switch: pin 12
  Encoder A: pin 8
  Encoder B: pin 9
Encoder #1
  Switch: pin 14
  Encoder A: pin 10
  Encoder B: pin 11
Encoder #2
  Switch: pin 17
  Encoder A: pin 2
  Encoder B: pin 3
Encoder #3
  Switch: pin 9
  Encoder A: pin 4
  Encoder B: pin 5
```

And this is my config:
```yaml
binary_sensor:
  - platform: gpio
    name: "BTN_UP"
    id: BTN_UP
    pin:
      number: GPIO6
      mode:
        input: true
        pullup: true
      inverted: true

  - platform: gpio
    name: "BTN_DOWN"
    id: BTN_DOWN
    pin:
      number: GPIO7
      mode:
        input: true
        pullup: true
      inverted: true

  - platform: seesaw
    id: button0
    name: "Seesaw encoder button #0"

  - platform: seesaw
    id: button1
    pin: 12
    name: "Seesaw encoder button #1"

  - platform: seesaw
    id: button2
    name: "Seesaw encoder button #2"

  - platform: seesaw
    id: button3
    name: "Seesaw encoder button #3"

```

```
INFO ESPHome 2024.3.2
INFO Reading configuration /config/esphome/adafruit-matrixportal-s3.yaml...
INFO Detected timezone 'Europe/Paris'
WARNING GPIO3 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO37 is used by the PSRAM interface on ESP32-S3R8 / ESP32-S3R8V and should be avoided on these models
WARNING GPIO45 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO36 is used by the PSRAM interface on ESP32-S3R8 / ESP32-S3R8V and should be avoided on these models
WARNING GPIO35 is used by the PSRAM interface on ESP32-S3R8 / ESP32-S3R8V and should be avoided on these models
Failed config

binary_sensor.seesaw: [source /config/esphome/adafruit-matrixportal-s3.yaml:170]
  platform: seesaw
  id: button1
  
  [pin] is an invalid option for [binary_sensor.seesaw]. Please check the indentation.
  pin: 12
  name: Seesaw encoder button #1
```

ssieb, 2024-04-08:
The `pin` parameter is only available in the `seesaw` **branch** where I'm working on it.

lboue, 2024-04-09:
Thanks . Now it works, but in reverse. The state is 'ON' when the button is not pressed.
[image]

closed 2024-04-16 by lboue