ssieb/esphome_components #73: [pin] is an invalid option for [sensor.seesaw]  and type: encoder

lboue, 2024-04-16:
Hello,

I am facing an issue with seesaw componant and my [Adafruit I2C Quad Rotary Encoder Breakout with NeoPixel - STEMMA QT / Qwiic, Product ID: 5752]( https://www.adafruit.com/product/5752). Encoders count in the wrong direction.

So I try to configure seesaw componant and set pins to change the behavior but I got this error:
`  [pin] is an invalid option for [binary_sensor.seesaw].` This is for encoder switch.

These are the **pin names** in the seesaw firmware for each rotary encoder coming from : [Adafruit I2C Quad Rotary Encoder Breakout tutorial](https://learn.adafruit.com/adafruit-i2c-quad-rotary-encoder-breakout/circuitpython-and-python)
```
Encoder #0
  Switch: pin 12
  Encoder A: pin 8
  Encoder B: pin 9
Encoder #1
  Switch: pin 14
  Encoder A: pin 10
  Encoder B: pin 11
Encoder #2
  Switch: pin 17
  Encoder A: pin 2
  Encoder B: pin 3
Encoder #3
  Switch: pin 9
  Encoder A: pin 4
  Encoder B: pin 5
```

And this is **my config**:
```yaml
external_components:
  - source:
      type: git
      url: https://github.com/ssieb/custom_components
      ref: seesaw
    components: [ seesaw ]

sensor:
  - platform: seesaw
    id: encoder0
    type: encoder
    number: 0
    #pin: 8 --> [pin] is an invalid option for [sensor.seesaw]
    name: "Seesaw encoder #0"

  - platform: seesaw
    id: encoder1
    type: encoder
    number: 1
    #pin: 10
    name: "Seesaw encoder #1"

  - platform: seesaw
    id: encoder2
    type: encoder
    number: 2
    #pin: 2
    name: "Seesaw encoder #2"

  - platform: seesaw
    id: encoder3
    number: 3
    type: encoder
    #pin: 4
```

I am getting this **log when I build** this config:
```
INFO ESPHome 2024.4.0b3
...
Failed config

sensor.seesaw: [source /config/esphome/adafruit-matrixportal-s3.yaml:345]
  platform: seesaw
  id: encoder0
  type: encoder
  number: 0
  
  [pin] is an invalid option for [sensor.seesaw]. Please check the indentation.
  pin: 8
  name: Seesaw encoder #0
sensor.seesaw: [source /config/esphome/adafruit-matrixportal-s3.yaml:352]
  platform: seesaw
  id: encoder1
  type: encoder
  number: 1
  
  [pin] is an invalid option for [sensor.seesaw]. Please check the indentation.
  pin: 10
  name: Seesaw encoder #1
sensor.seesaw: [source /config/esphome/adafruit-matrixportal-s3.yaml:359]
  platform: seesaw
  id: encoder2
  type: encoder
  number: 2
  
  [pin] is an invalid option for [sensor.seesaw]. Please check the indentation.
  pin: 2
  name: Seesaw encoder #2
sensor.seesaw: [source /config/esphome/adafruit-matrixportal-s3.yaml:366]
  platform: seesaw
  id: encoder3
  number: 3
  type: encoder
  
  [pin] is an invalid option for [sensor.seesaw]. Please check the indentation.
  pin: 4
  name: Seesaw encoder #3

```

Is there any way to correct this issue?

Regards

ssieb, 2024-04-16:
There's no way to specify the pins for the encoders.  Try the Arduino example sketch and see which way the numbers go.

lboue, 2024-04-16:
> There's no way to specify the pins for the encoders. Try the Arduino example sketch and see which way the numbers go.
@ssieb 
It works correctly with the Arduino sketch.

ssieb, 2024-04-16:
Yes.  I see I invert the values and there's a comment that it was to make clockwise positive.  Which way is positive with the arduino sketch?

lboue, 2024-04-16:
> Yes. I see I invert the values and there's a comment that it was to make clockwise positive. Which way is positive with the arduino sketch?

I get this in the logs during execution when turning encoders clockwise:
```
12:15:01.850 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
12:15:01.850 -> configsip: 188777542, SPIWP:0xee
12:15:01.850 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
12:15:01.850 -> mode:DIO, clock div:1
12:15:01.850 -> load:0x3fff0030,len:1416
12:15:01.850 -> load:0x40078000,len:14804
12:15:01.850 -> load:0x40080400,len:4
12:15:01.850 -> load:0x40080404,len:3356
12:15:01.850 -> entry 0x4008059c
12:15:01.979 -> Looking for seesaw!
12:15:02.026 -> seesaw started
12:15:02.026 -> Found Product 5752
12:15:02.073 -> Turning on interrupts
12:15:05.591 -> Encoder #0 -> 1
12:15:05.728 -> Encoder #0 -> 2
12:15:05.766 -> Encoder #0 -> 3
12:15:05.837 -> Encoder #0 -> 4
12:15:05.928 -> Encoder #0 -> 5
12:15:06.021 -> Encoder #0 -> 6
12:15:06.115 -> Encoder #0 -> 7
12:15:06.254 -> Encoder #0 -> 8
12:15:12.678 -> Encoder #0 -> 9
12:15:12.802 -> Encoder #0 -> 10
12:15:12.891 -> Encoder #0 -> 11
12:15:12.977 -> Encoder #0 -> 12
12:15:13.015 -> Encoder #0 -> 13
12:15:13.147 -> Encoder #0 -> 14
12:15:13.238 -> Encoder #0 -> 15
12:15:13.318 -> Encoder #0 -> 16
12:15:13.500 -> Encoder #0 -> 17
12:15:14.767 -> Encoder #1 -> 1
12:15:14.806 -> Encoder #1 -> 2
12:15:14.886 -> Encoder #1 -> 3
12:15:15.011 -> Encoder #1 -> 4
12:15:15.105 -> Encoder #1 -> 5
12:15:15.149 -> Encoder #1 -> 6
12:15:15.241 -> Encoder #1 -> 7
12:15:15.396 -> Encoder #1 -> 8
12:15:15.990 -> Encoder #2 -> 1
12:15:16.262 -> Encoder #2 -> 2
12:15:16.355 -> Encoder #2 -> 3
12:15:16.390 -> Encoder #2 -> 4
12:15:16.481 -> Encoder #2 -> 5
12:15:16.574 -> Encoder #2 -> 6
12:15:17.752 -> Encoder #3 -> 1
12:15:17.799 -> Encoder #3 -> 2
12:15:17.832 -> Encoder #3 -> 3
12:15:17.924 -> Encoder #3 -> 4
12:15:17.971 -> Encoder #3 -> 5
```