ssieb/esphome_components #74: Adafruit I2C Quad Rotary Encoder Breakout with NeoPixel / Seesaw Neopixel LED issue with 4 LED

lboue, 2024-04-16:
Hello,

I am facing an issue with seesaw componant and my [Adafruit I2C Quad Rotary Encoder Breakout with NeoPixel - STEMMA QT / Qwiic, Product ID: 5752](https://www.adafruit.com/product/5752). Encoders count in the wrong direction.

[image]

The LED on the first encoder works correctly, but the other 3 LEDs cannot be controlled.

I am using the `seesaw `branch of this github.

These are the **pin names** in the seesaw firmware for each rotary encoder coming from : [Adafruit I2C Quad Rotary Encoder Breakout tutorial](https://learn.adafruit.com/adafruit-i2c-quad-rotary-encoder-breakout/circuitpython-and-python)

```yaml
external_components:
  - source:
      type: git
      url: https://github.com/ssieb/custom_components
      ref: seesaw
    components: [ seesaw ]

light:
  - platform: seesaw
    id: pixel0
    pin: 18
    name: "Seesaw neopixel"
    #num_leds: 4

  - platform: seesaw
    id: pixel0
    pin: 18
    name: "Seesaw neopixel"
    #num_leds: 4
```

The `num_leds `parameter is not working:
```
Failed config

light.seesaw: [source /config/esphome/adafruit-matrixportal-s3.yaml:138]
  platform: seesaw
  id: pixel0
  pin: 18
  name: Seesaw neopixel
  
  [num_leds] is an invalid option for [light.seesaw]. Please check the indentation.
  num_leds: 4
```

ssieb, 2024-04-16:
Yes, that's under development.  I think I even mentioned to you that I'm figuring out how to implement that.  Please don't make more issues here, you can contact me on discord.  And you can't just make up undocumented parameters. :-)

m4rkireland, 2024-10-25:
Any updates on this one?

ssieb, 2024-10-25:
It's an RGB light.  You use the same methods that you normally use in esphome for a light.  You have to use a "call" as shown in the light docs.