ssieb/esphome_components #79: on_clockwise & on_anticlockwise triggers invalid options for [Adafruit StemmaQT Rotary Encoder]

zala143, 2024-05-31:
looking for some help.. I'm trying to use the [Adafruit StemmaQT Rotary Encoder] with ESPHome 2024.5.4; I've got most of the basic functionality working/responding to actions (rotary, button and the neopixel) - the part I cannot get working is the rotary encoder trigger that recognizes **on_clockwise** or **on_anticlockwise** triggers?  

kinda stuck if thats not supported - trying to use this sensor as a knob for a thermostat so need to simplify and know if dial is being turned in either direction.  I wanted to use simple wiring (I2C) so opted for this type of rotary encoder versus a regular 5 wire ver..

Any help would be greatly appreciated. 

Martin

ssieb, 2024-05-31:
That's not supported yet.  You can use the on_value and store the previous value in a static local variable or a global to compare against.

zala143, 2024-06-01:
ok many thanks, I guessed that might be the case - would be awesome for this feature to exist as it greatly simplifies using it within ESPHome. I will work out how to do as you said as I still feel the simplicity of the I2C wiring is my best route.. thank you