ssieb/esphome_components #82: Esmart3 doesn't show sensors

btmerz, 2024-08-21:
Hello,
I tried to implement my Esmart3 in Esphome an dtried to use your code.
I was succsessfully compiling and uploading the yaml file. Unfortunately I don't get any sensos shown.

Here is my code:

substitutions:
  name: esmart3
  device_description: "Esmart 3 MQTT Laderegler"
  external_components_source: https://github.com/ssieb/esphome_components
  tx_pin: GPIO17
  rx_pin: GPIO16
  ip: "122"   

esphome:
  name: ${name}
  comment: ${device_description}

esp32:
  board: lolin32_lite

external_components:
  - source:
      type: git
      url: ${external_components_source}
    components: [esmart3]

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:

web_server:
  port: 80    

logger:
  level: DEBUG

api:

uart:
  id: uart_1
  baud_rate: 9600
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}

sensor:
  - platform: esmart3
    charge_mode:
      id: cm
    input_voltage:
      id: iv
    battery_voltage:
      id: bv
    charging_current:
      id: cc
    load_voltage:
      id: lv
    load_current:
      id: lc
    charging_power:
      id: cp
    load_power:
      id: lp
    battery_temp:
      id: bt
    internal_temp:
      id: it
    battery_level:
      id: bl

Here ist the corresponding output of logger:
[22:10:21][C][logger:447]: Logger:
[22:10:21][C][logger:448]:   Level: DEBUG
[22:10:21][C][logger:449]:   Log Baud Rate: 115200
[22:10:21][C][uart.arduino_esp32:137]: UART Bus 1:
[22:10:21][C][uart.arduino_esp32:138]:   TX Pin: GPIO17
[22:10:22][C][uart.arduino_esp32:144]:   Data Bits: 8
[22:10:22][C][uart.arduino_esp32:145]:   Parity: NONE
[22:10:22][C][uart.arduino_esp32:146]:   Stop bits: 1
[22:10:22][C][esmart3:010]: Esmart3:
[22:10:22][C][esmart3:011]:   Charge Mode 'cm'
[22:10:22][C][esmart3:011]:     State Class: ''
[22:10:22][C][esmart3:011]:     Unit of Measurement: ''
[22:10:22][C][esmart3:011]:     Accuracy Decimals: 0
[22:10:22][C][esmart3:012]:   Input Voltage 'iv'
[22:10:22][C][esmart3:012]:     State Class: ''
[22:10:22][C][esmart3:012]:     Unit of Measurement: 'V'
[22:10:22][C][esmart3:012]:     Accuracy Decimals: 1
[22:10:22][C][esmart3:012]:     Icon: 'mdi:flash'
[22:10:22][C][esmart3:013]:   Battery Voltage 'bv'
[22:10:22][C][esmart3:013]:     State Class: ''
[22:10:22][C][esmart3:013]:     Unit of Measurement: 'V'
[22:10:22][C][esmart3:013]:     Accuracy Decimals: 1
[22:10:22][C][esmart3:013]:     Icon: 'mdi:flash'
[22:10:22][C][esmart3:014]:   Charging Current 'cc'
[22:10:22][C][esmart3:014]:     State Class: ''
[22:10:22][C][esmart3:014]:     Unit of Measurement: 'A'
[22:10:22][C][esmart3:014]:     Accuracy Decimals: 1
[22:10:22][C][esmart3:014]:     Icon: 'mdi:current-ac'
[22:10:22][C][esmart3:015]:   Load Voltage 'lv'
[22:10:22][C][esmart3:015]:     State Class: ''
[22:10:22][C][esmart3:015]:     Unit of Measurement: 'V'
[22:10:22][C][esmart3:015]:     Accuracy Decimals: 1
[22:10:22][C][esmart3:015]:     Icon: 'mdi:flash'
[22:10:22][C][esmart3:016]:   Load Current 'lc'
[22:10:22][C][esmart3:016]:     State Class: ''
[22:10:22][C][esmart3:016]:     Unit of Measurement: 'A'
[22:10:22][C][esmart3:016]:     Accuracy Decimals: 1
[22:10:22][C][esmart3:016]:     Icon: 'mdi:current-ac'
[22:10:22][C][esmart3:017]:   Charging Power 'cp'
[22:10:22][C][esmart3:017]:     State Class: ''
[22:10:22][C][esmart3:017]:     Unit of Measurement: 'W'
[22:10:22][C][esmart3:017]:     Accuracy Decimals: 0
[22:10:22][C][esmart3:017]:     Icon: 'mdi:power'
[22:10:22][C][esmart3:018]:   Load Power 'lp'
[22:10:22][C][esmart3:018]:     State Class: ''
[22:10:22][C][esmart3:018]:     Unit of Measurement: 'W'
[22:10:22][C][esmart3:018]:     Accuracy Decimals: 0
[22:10:22][C][esmart3:018]:     Icon: 'mdi:power'
[22:10:22][C][esmart3:019]:   Battery Temp 'bt'
[22:10:22][C][esmart3:019]:     State Class: ''
[22:10:22][C][esmart3:019]:     Unit of Measurement: '°C'
[22:10:22][C][esmart3:019]:     Accuracy Decimals: 0
[22:10:22][C][esmart3:019]:     Icon: 'mdi:thermometer'
[22:10:22][C][esmart3:020]:   Internal Temp 'it'
[22:10:22][C][esmart3:020]:     State Class: ''
[22:10:22][C][esmart3:020]:     Unit of Measurement: '°C'
[22:10:22][C][esmart3:020]:     Accuracy Decimals: 0
[22:10:22][C][esmart3:020]:     Icon: 'mdi:thermometer'
[22:10:22][C][esmart3:021]:   Battery Level 'bl'
[22:10:22][C][esmart3:021]:     State Class: ''
[22:10:22][C][esmart3:021]:     Unit of Measurement: '%'
[22:10:22][C][esmart3:021]:     Accuracy Decimals: 0
[22:10:22][C][esmart3:021]:     Icon: 'mdi:percent'
[22:10:22][D][api:102]: Accepted 192.168.178.6
[22:10:22][D][api.connection:1121]: Home Assistant 2024.8.1 (192.168.178.6): Connected successfully
[22:10:22][C][web_server:168]: Web Server:
[22:10:22][C][web_server:169]:   Address: 192.168.178.122:80
[22:10:22][C][mdns:115]: mDNS:
[22:10:22][C][mdns:116]:   Hostname: esmart3
[22:10:22][C][ota:096]: Over-The-Air Updates:
[22:10:22][C][ota:097]:   Address: 192.168.178.122:3232
[22:10:22][C][ota:103]:   OTA version: 2.
[22:10:22][C][api:139]: API Server:
[22:10:22][C][api:140]:   Address: 192.168.178.122:6053
[22:10:22][C][api:144]:   Using noise encryption: NO
[22:11:09][D][esp32.preferences:114]: Saving 1 preferences to flash...
[22:11:09][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed

Here is the device in homeassistant:
[image]

Perhaps you can help me out here...

randybb, 2024-08-21:
Looks ok to me as you have only `id:` under your sensors, which means you don't want to expose them to HA. Otherwise you would use `name:`.

ssieb, 2024-08-21:
As with any esphome sensor, you have to give it a `name:` or it's internal only.

closed 2024-08-21 by ssieb

btmerz, 2024-08-21:
Thanks for your help. Now I can see the sensors
Perhaps it would be helpful, if you include that in the readme file, as I copied the code from there...

Unfortunately I don't get any sensor-values. Maybe my RS485 module is faulty. I will check later...
Thanks again.

ssieb, 2024-08-21:
That's standard esphome config behaviour.  However, in newer examples I have been putting names instead of ids, so I might update that at some time.

btmerz, 2024-09-04:
Hello Samuel,
I now have ordered a new uart - RS485 module and the ESPhome firmware running. Unfortunately I don't receive any sensor data from the Esmart.
The RX / TX LEDs doesn't flash also. Pin 1 is on A and Pin 2 is on B.
Do you have any advice?

ssieb, 2024-09-04:
Come find me on the esphome discord server.  It will be easier to help there.