ssieb/esphome_components #83: eta_sh get no data at all

joe16joe16, 2024-11-07:
I use a ESP32 and this adapter (https://www.amazon.de/dp/B0BNQ41QKQ) and yaml:

`########################################################################
substitutions: #########################################################
  devicename: "esp32-eta"
  fn: Heizung

esphome:
  name: $devicename
  friendly_name: $fn

esp32: #################################################################
  board: nodemcu-32s
#  framework:
#    type: arduino
#    version: 2.0.6

# Enable logging
logger:
#  level: DEBUG
#  level: VERY_VERBOSE
#  baud_rate: 0
# Enable Home Assistant API

api:
  encryption:
    key: !secret has_key

ota:
  - platform: esphome  
    password: !secret ota_password

web_server:
  version: 2

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  reboot_timeout: 60 min  # reboot without wifi only every thirty minutes --> automation needs to work stand alone too!
  fast_connect: true
  use_address: $devicename.home.kienzer.com
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${devicename}" Fallback Hotspot"
    password: !secret fallback_password

captive_portal:
    
external_components:
  - source:
      type: git
      url: https://github.com/ssieb/esphome_components
    components: [ eta_sh ]

uart:
  rx_pin: GPIO26
  tx_pin: GPIO22
  baud_rate: 19200 
  stop_bits: 1
  data_bits: 8
  parity: NONE
#  rx_buffer_size: 512

eta_sh:
  boiler_temperature:
    name: Kesseltemperatur  #(Boiler temp)
  fan_speed:
    name: Gebläsedrehzahl   #(Speed of boiler Fan)
    unit_of_measurement: RPM
  return_temperature:
    name: Kesselruecklauf   #(Boiler water return temp)
  buffer_bottom_temperature:
    name: Puffer unten      #(Water buffer bottom temp)
  buffer_middle_temperature:
    name: Puffer mitte      #(Water buffer middle temp)
  buffer_top_temperature:
    name: Puffer oben       #(Water buffer top temp)
  exhaust_temperature:
    name: Abgastemperatur   #(Exhaust temp)
  room1_temperature:
    name: Raumtemperatur    #(Room1 temp)
  room1_output_temperature:
    name: Vorlauf Raum 1    #(Water to room1 temp)
  outside_temperature:
    name: Aussentemperatur  #(Outside temp)
  buffer_load:
    name: Pufferladezustand #(Buffer load in %)
  external_heater_temperature:
    name: Externer Brenner  #(External Oven (expansion, e.g. Oil or Gas)
  oxygen_sensor:
    name: "Restsauerstoff"   	#(o2 Sensor Percent / Lambda Sond. Returns 0 when oven off)
  heater_status:
    name: "Status"          #            Status of the heater,
                            #            0: "Tür offen"			(Door open)
                            #            1: "heizt an"			(starting Fire)
                            #            2: "in Betrieb"			(working)
                            #            3: "Übertemperatur"		(over temperature Limit)
                            #            4: "Glutabbrand"		(embers burning)
                            #            5: "Feuer aus"			(no Fire)
                            #            6: "Ausgeschaltet"		(heater off)
                            #            7: "Störung"			(fault)
`

But when I connect it to the display serial interface of the eta sh 20 I get no data at all.

In the past I use a raspberry with this C daemon:
http://ulrich-franzke.de/haustechnik/eta_programm1.html

Do I miss something in the yaml or is your code for another version?

Many thank's for your reply in advance

ssieb, 2024-11-08:
Try swapping the RX and TX pins.  If you're still having issues, come find me on the esphome discord server.

ssieb, 2024-11-11:
The dummy receiver will stop the component from working.  You only use that if there isn't a component already listening.